keyboardio/Kaleidoscope #882: Mix LED-ActiveModColor with QuKeys

BenjaminGalliot, 2020-07-21:
Hello,

It seems there is no way for now to make LED-ActiveModColor work on a Qukey like RALT_T(PcApplication) (PcApp is my Compose key on tap and Alt-Gr on hold), in order to let the holding Alt-Gr be colored like an Alt or a Ctrl.

Sincerely.

gedankenexperimenter, 2020-08-07:
This could probably be fixed by changing `Layer.lookupOnActiveLayer()` to `Layer.lookup()` in `LED-ActiveModColor.cpp`, since Qukeys alters the live composite keymap state without changing the value in the active layer(s) of the keymap. I haven't tested this, but I'm pretty sure it wouldn't affect OneShot keys.

gedankenexperimenter, 2020-08-30:
I'm looking at LED-ActiveModColor in more detail, to see if my idea would work, and it won't be sufficient. The plugin builds a static list of key addresses that correspond to modifiers whenever layer state changes. It could be expanded to include DualUse Qukeys as well as OneShot keys. It could even be expanded to include full Qukeys, but here's the problem: the array in which they're stored is limited to 16 elements, and there's no bounds-checking on that array, so if there are more than 16 (potential) modifier keys in the current keymap, it can write past the end of the array.

I've successfully used bitmaps in a few places in place of arrays like this one, and there's an efficient way of searching them (bytes first, then bits) when most bits will be zeroes most of the time. On a Model01, with 64 keys, this bitmap of active modifier state would take only half as much memory as the current code, and would protect against writing memory out of bounds, too.

Furthermore, with this system, the plugin can be expanded to deal with all Qukeys, and even OneShot keys that are injected from Macros (or other sources), by using a key event handler hook. I might take on this project if I get some free time in the next week.

gedankenexperimenter, 2020-09-02:
@BenjaminGalliot — I've submitted a PR (#890) that makes ActiveModColor compatible with Qukeys, as well as OneShot and plain modifiers.