keyboardio/Kaleidoscope #884: Remove "key masking"?

obra, 2020-08-20:
@algernon mentioned that we may no longer need the key masking abstraction. If that's true, it should go away.

algernon, 2020-08-24:
The original reason why masking was introduced (in 59bc052301be6ce395e2ceea11604071b99aed9f) was for OneShots, to avoid the case of having unintentional keys sent. As in, if we had a OneShot active, and we pressed a key that'd interrupt it, the expected outcome was to have OneShot activate for that key only. But even a short tap usually results in the key being held for multiple cycles, so we ended up in a situation where we had the key with oneshot, and then without. To avoid this situation, we masked the key until release. For reference, masking means that key events are ignored until the key is released: this way, we don't send extras, because we ignore the key until released.

However, since then, we changed how keys are looked up, and such masking became unnecessary, and with #434, the masking was dropped from OneShot.

At the moment, `maskKey` is used in two places: `Escape-OneShot` (introduced with 8f4a34d4b9a29300059e0d4bab029d4af433744a), and `TapDance` (introduced in a31d686911e4fdf0d3dedd1a456be2121edf93b9). Both of these appear to solve similar problems we had with OneShot, so the workarounds feel similarly unnecessary by this point.

I will verify shortly whether my assumptions are correct, and if so, we can remove the use of masking, and deprecate those interfaces.

algernon, 2020-08-24:
To clarify: the problem originally was that we had _different_ events for the same physical key, and _that_ resulted in problems. With the key caching thing we have now, this is no longer the case for OneShot, nor should it be the case for TapDance.

We might still need it for `Escape-OneShot`, though, because even though the root problem is the same there, caching doesn't solve it. In case of Escape-OneShot, the idea is that if OneShot is active, pressing Esc will cancel it - and we will not send an Esc in the report. We currently do this by masking the key. If we remove the masking, then hitting Esc will cancel oneshot in the next cycle. Chances are, we won't be releasing it fast enough, and we'll have another cycle with Esc still pressed. At that point, OneShot is no longer active, so the plugin just lets it through, and `Esc` will end up getting added to the report.

So we need masking there still, or we change how the plugin works... if we stop consuming the event, and instead change the mapped key to `Key_NoKey` in case of a cancel, then we'll benefit from caching, and might be able to drop the masking.

obra, 2020-08-24:
I'd love it if you'd explore a change to Escape-OneShot. Removing key
masking would save us a bit of size and complexity.

algernon, 2020-08-24:
We can't immediately remove it, without deprecating it first, but I get you. PR to change Escape-OneShot will be up later tonight, and then another PR that deprecates masking.

commit 8d4dfa87 (2020-08-25): Deprecate the key masking API

Key masking was a bandaid, and we have better ways to achieve the same thing
now. All current users have been switched over to different methods now, so lets
deprecate the masking.

We only put the `DEPRECATED` label on the `maskKey` method, because the rest are
used internally too, and we do not want to emit warnings for those.

Fixes #884.
closed 2020-08-25 by obra