keyboardio/Kaleidoscope #910: Raising active layers

algernon, 2020-09-28:
This is a followup for #857 and #867, and assumes that #867 has been accepted.

One interesting point raised on Discord by @gedankenexperimenter with regards to activation order is the question of what should happen if we re-activate an already active layer, which wasn't on top? The current behaviour is to do nothing, because the layer is already active. The proposed change would be to bring the re-activated layer to the top. So if we activate 1, 2, and then 1 again, 1 would be on top now.

algernon, 2020-09-28:
There are, however, a number of complicating factors, mostly related to locked layers: at the moment, `LockLayer` and `UnlockLayer` are exactly the same keycode, they toggle the layer on and off. Thus, the only way to activate the same layer twice is to use either a macro, or a combination of different types of layer keys - neither of which sounds realistic. If we go down this route, and allow raising an active layer, then we need to separate `LockLayer` and `UnlockLayer` - that would be a breaking change, because existing keymaps would no longer work as they did before.

To combat that, we'd need to introduce a new pair of keycodes for the new LockLayer/UnlockLayer behaviour, and still keep the old one around. This would make the logic quite complicated and messy. And we didn't talk about plugins yet, which may complicate things further.

To avoid that mess, we could introduce another layer key: ActivateLayer (or RaiseLayer, or ReactivateLayer, ...) which would be a different keycode than LockLayer/UnlockLayer, and would simply bring the selected layer to the top. Deactivating it would still work with the existing `UnlockLayer` key. This avoids the compatibility mess, at the cost of having one more kind of layer key out of the box.

We already have LockLayer/UnlockLayer, ShiftToLayer, OneShots, DualUse via Qukeys, and macros...

Come to think of it... it's already possible to bring a layer to the top via macros, with #867 in place: deactivate it, and then re-activate it, and boom, it's on top. With a macro, that can be done in a single step, without any other events happening inbetween (unless there's a plugin that listens to OnLayerChange events and does something funky).

algernon, 2020-09-28:
To sum it up: I see the use of such functionality, but I would not change the existing behaviour - partly because that wouldn't be possible without breaking LockLayer/UnlockLayer apart. I'd rather add an extra key that brings a layer to the top - that would be very simple to do, and wouldn't present any compatibility issues.

gedankenexperimenter, 2020-09-28:
Like @obra said on Discord, let's not get too hung up on names, some of which were questionable in the first place (i.e. `LockLayer`/`UnlockLayer`), and focus instead on what useful behaviours we could have from layer keys.

I think there are two basic categories of layer key: ones that make a keymap change while the key is held (and undo that change on release), and ones that respond to taps (and leave that change in effect after release). I'll call these two types of keys “layer-hold” and “layer-tap” keys, respectively.

1. I think it would be best to have one default layer-hold key, and one default layer-tap key, each with an index parameter.

2. I think that, whatever the behaviour of these default keys, `ShiftToLayer(N)` and `LockLayer(N)` should become aliases for whatever new names are decided on, because these are the keys that already appear in peoples' keymaps. Conceptually, these two keys do the most sensible thing for layer-hold and layer-tap keys given simple index ordering. They should therefore continue to do whatever is most sensible for layer-hold and layer-tap keys if the keymap switches to activation-order. `UnlockLayer(N)` could do something different, however; it need not continue to be an alias for `LockLayer(N)`.

3. I think the default layer-tap keys should guarantee causing a user-visible change to the keymap. The most straightforward way to do that is to bring the target layer to the top if it's not there, and deactivate it if it is there. This pseudo-toggle behaviour should be the least surprising behaviour, except perhaps in cases where the target layer has keys that are currently visible because of a higher layer's transparent keys.

4. I submit that we should strive in the design to not require the user to keep track of the current layer stack state. The user must be able to keep track of the current _keymap_, but if the layers are (effectively) all non-transparent, all they should need to know is “what layer am I on now?” and “what layer will I return to when this one deactivates?”

5. The default Model01 keymap has two (mostly) complete layers, and one substantially transparent one. I don't know if this is common for other keyboards, or for custom keymaps (or how prevalent custom keymaps are, for that matter), but a big question we should be asking is: “On the default Model01 keymap, what should happen if the `NUMPAD` layer is on top, and the `FUNCTION` layer is active, below it, and we press and release `ShiftToLayer(FUNCTION)`?” While that layer-hold key (I wrote `ShiftToLayer` because that's what's in the default keymap now) is being held, should the `FUNCTION` layer move to the top, above `NUMPAD`? When it's released, should the `FUNCTION` layer be deactivated, or should it be restored to its prior position?

6. Maybe layer-hold keys should always take priority over layer-tap (lock) keys.  Suppose (again, in the default Model01 keymap), I start from the base layer (only), press and hold one palm key, then tap the `LockLayer(NUMPAD)` key. Since I'm holding the palm key, maybe it would be less surprising to me if the `FUNCTION` layer stayed on top. If not, and `NUMPAD` becomes the top layer, then suppose I press and hold the other palm key, then release one of the palm keys…

I'm rambling now, because I've got so many of these scenarios in my head, but there's one more I want to call out before I forget, which is somewhat unrelated: The default Model01 keymap has transparent keys on every layer except the base layer for the palm keys. If a small change is made to the keymap, it would be possible (I think) in the current design to have only the `FUNCTION` layer active, and not the base layer. If the layer stack is searched for a value for a `KeyAddr`, and it finds `Key_Transparent`, it should return whatever the value is for that entry on the base layer.