keyboardio/Kaleidoscope #922: TapDance interruption within (short) time_out generate wrong output

Kompost14, 2020-10-10:
Hello

i've an issue when a TabDance is interrupted within the Timeout, the output isn't what expected.

Coding:

```
/* Depends on library https://github.com/keyboardio/Kaleidoscope-Languages */
#include "Kaleidoscope-Languages.h"
#include "kaleidoscope/lang/de-qwertz.h"

#define TD_a  0 // a or ä

KEYMAPS(
   TD(TD_a)
)

KALEIDOSCOPE_INIT_PLUGINS(
  TapDance,
);

void tapDanceAction(uint8_t tap_dance_index, KeyAddr key_addr, uint8_t tap_count, kaleidoscope::plugin::TapDance::ActionType tap_dance_action) {
  switch (tap_dance_index) {
    case TD_a:
      return tapDanceActionKeys(tap_count, tap_dance_action, Key_de_A, Key_de_AUml);
  }
}

void setup() {
  TapDance.time_out = 180; // default value is 200
  Kaleidoscope.setup();
}

```
When now keys 'a''s' are fast tapped withn the time_out (180ms), the output ist 'asa' or 'ss'. Expected is 'as', that the TapDance is interrupted with the time_out.

There must be perhaps another time_counter/out. When the time_out ist set to an extrem value (e.g. 2000ms), an the time_out of a TapDance is interrupted after 500ms, the output is as expected correct. But when the TapDance is interrupted within the first 0 - <100ms(?), the output is incorrect.

gedankenexperimenter, 2020-10-10:
I suspect this only happens during rollover, when the second TapDance key is pressed before the first one is released. I will test it when I get a chance.

gedankenexperimenter, 2020-10-10:
@Kompost14 — You could try tapping the two keys quickly, but use the same finger for both to make sure you don't get any rollover, and see what happens.

gedankenexperimenter, 2020-10-11:
I can reproduce parts of this one. If I roll over from one TapDance key to another repeatedly, I will get alternating results:
- `TD(0)` => `(Key_A, Key_X)`
- `TD(1)` => `(Key_B, Key_Y)`

Rollover from `TD(0)` to `TD(1)` produces `ab`, then `y`, then `ab`, then `y`, and so on.

Rollover from `TD(1)` to `TD(0)` produces `bba`, then `x`, alternating. I used the same two keys, so this also seems to be scan-order dependent.

I haven't found a way to get `aba` or `bab` yet. Tapping the two keys without rollover, but still within the Tapdance timeout doesn't seem to result in any unexpected output.

If I roll over one way, then the other on these two keys, I get even more extra characters: `ayy`, then `bbxx`, and so on. The last character appears after a delay corresponding to the length of the TapDance timeout.

gedankenexperimenter, 2020-10-11:
Note: I strongly urge everyone to not set the TapDance timeout lower than the default, or to hold down any TapDance key longer than that timeout, because of #806.

I do have plans to rewrite TapDance to make it more robust, but it will take some time to get it done, and even more to get it merged.

gedankenexperimenter, 2020-11-19:
I now have a testcase for this one, and a fix in a forthcoming version of #904.

commit a956c2d7 (2021-05-28): Adapt TapDance to keyboard state array

This is a major rewrite of TapDance, taking advantage of the keyboard state
array and the `KeyAddrEventQueue` class originally written for Qukeys.

fixes #806
fixes #922
fixes #908
fixes #985
closed 2021-05-28 by algernon