keyboardio/Kaleidoscope #939: Test simulator: two `RunCycle()` functions confusing

gedankenexperimenter, 2020-10-19:
As I stumble my way through figuring out how to write tests using the simulator, I'm generally finding that I can figure out what I need to do, based on the tests already present. I have run into a few minor problems, however. In particular, there are two different functions named `RunCycle()`:
- `SimHarness::RunCycle()`, which calls `Kaleidoscope.loop()` one time to run through a whole scan cycle.
- `VirtualDeviceTest::RunCycle()`, which calls the above function, then returns a `State` value.
I found this quite confusing for a bit, probably because the testcase classes are derived from `VirtualDeviceTest`, so they call `RunCycle()` without needing to specify a namespace or class, and because I had been looking at `SimHarness` first, because the function that I wanted to call was `RunCycles(n)` to run through many cycles, not just one.

I guess I was sloppy, because once I had seen the `RunCycle()` function in `SimHarness.h`, and a function with that name being called from a testcase function, I assumed that it was the same function, and was surprised when it didn't work to call `RunCycles(n)` in the same manner (`sim_.RunCycles(n)` does work).

Maybe `VirtualDeviceTest::RunCycle()` should be renamed to `RunCycleAndGetState()` or something to that effect to improve clarity.

obra, 2020-10-19:
Thanks. The potential confusion is a valid point. I think my preference is to rename the one in SimHarness to something more expressive.

Let me ponder what name might make sense for it.

gedankenexperimenter, 2020-10-19:
My two cents: `RunCycle()` doesn't suggest to me that it would have a return value, and if you told me that it does, I couldn't guess what it would be.

gedankenexperimenter, 2020-11-18:
I think this is mostly a non-issue now that we have ktest, so I'm going to close it.

closed 2020-11-18 by gedankenexperimenter