keyboardio/Kaleidoscope #940: Test simulator: Add a way to verify a whole keyboard HID report

gedankenexperimenter, 2020-10-19:
When writing tests, I generally want to compare an entire keyboard HID report with an expected state, not just check that it contains a particular value (or doesn't). I would try to add this functionality myself, but I'm a bit lost amidst the preprocessor macros, and I doubt I'd get it right.

Basically, it's not clear to me how to compare two vectors of keycode values in one of the `EXPECT*()` macro functions. I _think_ the right thing to do is to add something to `matchers.h`, but I'm feeling too stupid to figure it out myself.

obra, 2020-10-19:
My one concern around that functionality would be that it can often result in very brittle tests, especially if any kind of assumption about the size or shape of the report being tested is baked into the test values.

gedankenexperimenter, 2020-10-19:
What I want is a way to specify a complete list of the keycodes that should be in a given report. There's already a way to extract a vector from the report without worrying about the way that it's stored in Kaleidoscope. If the testing infrastructure doesn't supply it, I'll just have to supply it myself somehow, and that will certainly be more brittle.

The values being checked are the essence of the test. To meaningfully do the tests I've got in mind, I need to check that a given report contains a particular set of keycodes, and no others. Maybe I'm misunderstanding your concern?

gedankenexperimenter, 2020-10-19:
Let me put it another way: `KeyboardReport::ActiveKeycodes()` returns a `std::vector<uint8_t>`. I need a way to compare that to my own `std::vector<uint8_t>` to confirm that they contain the same set of values, within one of the `EXPECT*()` macros. Otherwise, I basically can't write any meaningful tests.

obra, 2020-10-19:
And putting it that way, I think that we're supposed to already have enough rope for it, coming from gmock, but @paniag may have more input here.

https://github.com/google/googletest/blob/master/googlemock/docs/cheat_sheet.md

I -think- we get ElementsAre for free and that it's what you want.

gedankenexperimenter, 2020-10-19:
I'll see if I can figure it out tonight. Thanks!

gedankenexperimenter, 2020-10-20:
It's not exactly what I was hoping for, but `::testing::ElementsAreArray()` is close enough. I wanted something that wouldn't care about the order of the elements, but that should be okay, at least for NKRO.

closed 2020-10-20 by gedankenexperimenter