keyboardio/Kaleidoscope #941: Test simulator: Fails to process two key presses in the same cycle

gedankenexperimenter, 2020-10-20:
I may be doing something wrong, but when I write a testcase that sends two "simultaneous" keypresses (`sim_.Press(x, y)`) before calling `RunCycle()`, only one of them ends up in the ensuing report.

PR with failing testcase coming shortly.

gedankenexperimenter, 2020-10-20:
Now that I've played around with this testcase, I suspect that it's just that I don't understand the test simulator. In Kaleidoscope, if two ordinary keyboard keys toggle on in the same cycle, I expect that to generate just one keyboard HID report. In the simulator, however, I get two elements in `RunCycle()->HIDReports()->Keyboard()`. The first contains only one keycode (the one processed first in scan order), and the second contains both. What is the purpose of this first (bogus) report? This looks to me like we're not seeing actual HID reports that are sent to the host, but in-progress HID report states built during the scan cycle. I suppose this is useful for debugging scan-order issues, but I'm primarily interested in the output to the host, not the intermediate states, which could change without affecting the output. Do we have a method for accessing only reports that are actually sent?

gedankenexperimenter, 2020-10-20:
I'm now mostly convinced that my second analysis was wrong, and that it's the virtual hardware device that's producing multiple reports. I can't find the code that gets run for the virtual device, though, so I'm stuck.

obra, 2020-10-21:
https://github.com/keyboardio/Kaleidoscope/blob/master/src/kaleidoscope/driver/hid/base/Keyboard.h#L179 is the code path that's causing a second HID report for two keypresses in the same cycle.

It was originally https://github.com/keyboardio/Kaleidoscope-HIDAdaptor-KeyboardioHID/commit/7b91aec54900d63390e4226a997213a19195355a  which itself reverted https://github.com/keyboardio/Kaleidoscope-HIDAdaptor-KeyboardioHID/commit/ff727ee5efbed8e052cfcde12df1d0888840a803

I believe the original code was added in: a4368f13e7a1b58ef7212147f4f0e1ced131ff75

obra, 2020-10-21:
Reworked the sample sketch for the test and merged to master.

closed 2020-10-21 by obra

gedankenexperimenter, 2020-10-21:
Explanation in #945, but I'll summarize here, as well. This could be considered a bug, but I wouldn't. The extra report takes some time, but there are circumstances in which it would be necessary to send even more than one extra report (and more than Kaleidoscope does send) to really get it right, when keyboard keys with modifier flags are involved.