keyboardio/Kaleidoscope #951: A subtle issue with layer states

algernon, 2020-10-28:
When trying to figure out what happened in keyboardio/Chrysalis#599, I stumbled upon a subtle bug in the layer state handling code. When Kaleidoscope starts up, we set `active_layer_count_` to `1`, and clear `active_layers_` and `layer_state_`. In practice, this means that layer 0 will both be active and not. It will be active as far as lookup is concerned, because the active layer count is 0, and the first entry in `active_layers_` is 0. However, `isActive(0)` will report false, because `layer_state_` is completely clear, and we do _not_ flip layer 0's bit open.

In practice, this shouldn't cause any issues, but it's inconsistent. We should be setting up `layer_state_` properly in the beginning too. One way to do that would be an explicit activation, but that would invoke the `onLayerChange()` hook, which we do not want at startup. We could have a `Layer::setup()` method that does any initial setup, including flipping a bit on. We could also have `updateActiveLayers()` recompute `layer_state_` too.

I'm leaning towards `Layer::setup()`, because that's more future-proof. We have a setup method for most other components of the firmware, we should have one for layers too. That would remove the slightly awkward keymap cache update from `Runtime::setup()`, as it would get replaced by `Layer.setup()` instead.

obra, 2020-10-28:
Can we start with a failing test?

algernon, 2020-10-28:
I _think_ I can even write a test for this, if I'm a bit clever. Since we want to test `Layer.isActive(0)`, and we can't reach that directly from the tests, we'll use a macro in the test sketch, that does different things based on whether the function returns true or false. Then we can examine the report after.

So yep, we can start with a failing test.

gedankenexperimenter, 2020-10-28:
Are you sure you can't reach that directly from the testcase?

gedankenexperimenter, 2020-10-28:
I just tried it out, and I can write a test for this, quite simply, without resorting to Macros.

commit 0cfc546a (2020-10-28): Layers: Fix the initial state of layer 0

When starting up, we correctly set the active layer counter to one, and the
active layer stack will therefore correctly contain layer 0 as an active layer.
However, we weren't setting the `layer_state_` bitmap up properly, and as such,
`Layer.isActive(0)` was returning false, despite the layer being active as far
as lookups were concerned.

To fix this, we explicitly flip the 0th bit on in the newly introduced
`Layer.setup()` method, where the initial keymap cache update was moved to, too.

Fixes #951.
commit 60b5b24b (2020-10-28): Layers: Fix the initial state of layer 0

When starting up, we correctly set the active layer counter to one, and the
active layer stack will therefore correctly contain layer 0 as an active layer.
However, we weren't setting the `layer_state_` bitmap up properly, and as such,
`Layer.isActive(0)` was returning false, despite the layer being active as far
as lookups were concerned.

To fix this, we explicitly flip the 0th bit on in the newly introduced
`Layer.setup()` method, where the initial keymap cache update was moved to, too.

Fixes #951.
closed 2020-11-05 by obra