keyboardio/Kaleidoscope #980: TapDance allows additional keypresses through before executing

WillSoper, 2020-11-17:
When using the TapDance plugin, it sometimes passes keys through in an order that is not expected. 

Take the following example, where the Left Control key is to be remapped to MoveToLayer(1) when tapped twice:
```
void tapDanceAction(uint8_t tap_dance_index, KeyAddr key_addr, uint8_t tap_count, kaleidoscope::plugin::TapDance::ActionType tap_dance_action) {
  switch (tap_dance_index) {
  case 0:
    return tapDanceActionKeys(tap_count, tap_dance_action, Key_LeftControl, MoveToLayer(1));
  }
}
```
If a key on layer 0 is mapped to this TapDance action in Chrysalis, then that key is tapped twice quickly (to activate layer 1) then the next key that is pressed sometimes sends the key from its binding on Layer 0 next instead of Layer 1. The keypress **after** that will correctly report as being from Layer 1, as though the keys are being processed out of order. 

This issue is not 100% reproducable (it seems intermittent) but I see it roughly 50% of the time (using an Atreus).

More context here, where this was originally discussed: 
https://community.keyboard.io/t/tapdance-on-ctrl-key-can-i-movetolayer/4354/4

gedankenexperimenter, 2020-11-17:
I've just been working on TapDance today, to adapt it for #904. I think I'll be able to write a testcase that will reproduce this issue reliably, and I also expect that the changes I'm working on will fix it.

gedankenexperimenter, 2020-11-18:
I can reliably reproduce this in a testcase. The problem occurs in rollover from the TapDance key to the other key; if you make sure to release the TapDance key before pressing the other key, it works. I can also confirm that a non-yet-pushed version of #904 passes that test.

closed 2021-05-28 by algernon