keyboardio/Kaleidoscope #983: Kickstarter Keyboardio Atreus can control neither EFI boot menus nor GRUB.

parke, 2020-11-21:
My computer is a Chromebox, flashed with a Coreboot EFI firmware as provided by mrchromebox.tech.  The EFI boot menu does not respond to keypresses on my Atreus.  Nor does GRUB respond to Atreus keypresses.  The Atreus only becomes functional after the Linux kernel has booted.

I have not rigorously tested the Atreus with other systems, so I don't know how specific this problem is to the Atreus (or to the Chromebox).  I have verified that my Ultimate Hacking Keyboard can control both EFI and GRUB on the Chromebox.

gedankenexperimenter, 2020-11-22:
This is probably happening because it's in NKRO mode, and the host is assuming 6KRO ("Boot") mode without properly requesting it. There is a way to toggle the HID protocol Kaleidoscope uses, but it might not be included in the default Atreus sketch, like it is in the Model01-Firmware repository.

rmlrml, 2020-12-18:
I'm having the exact same problem on my Librem Mini.