keyboardio/Kaleidoscope #991: Qukeys examples has issues with arduino

bolob, 2020-12-06:
in the examples of qukeys : the keyAddr are wrong

in the arduino version, there is no Qukeys.setMinimumPriorInterval() method.

gedankenexperimenter, 2020-12-06:
Can you elaborate on this? Which `KeyAddr` is wrong?

Re: `setMinimumPriorInterval()`: This is the most recent addition on the master branch. I guess the current boards-manager version is not up to date. The [previous version](https://github.com/keyboardio/Kaleidoscope/blob/9b70161a9b1d48735ae6fec3f6881a95f3e9fea5/examples/Keystrokes/Qukeys/Qukeys.ino) of the example should work.

Rom1deTroyes, 2020-12-06:
With actual master branch (34bb24a7f1c60bbd393b8b6c1b5480a3bc93d2fb refs/heads/master), `src/kaleidoscope/plugin/Qukeys.h` line 135:

```cpp
  // Set the minimum interval between the previous keypress and the qukey press
  // to make the qukey eligible to become its alternate keycode.
  void setMinimumPriorInterval(uint8_t min_interval) {
    minimum_prior_interval_ = min_interval;
  }
```

bolob, 2020-12-06:
the following lines are not correct : 
    kaleidoscope::plugin::Qukey(0, KeyAddr(2, 1), Key_LeftGui),      // A/cmd
    kaleidoscope::plugin::Qukey(0, KeyAddr(2, 2), Key_LeftAlt),      // S/alt
    kaleidoscope::plugin::Qukey(0, KeyAddr(2, 3), Key_LeftControl),  // D/ctrl
    kaleidoscope::plugin::Qukey(0, KeyAddr(2, 4), Key_LeftShift),    // F/shift
    kaleidoscope::plugin::Qukey(0, KeyAddr(3, 6), ShiftToLayer(1))   // Q/layer-shift (on `fn`)

Those key addresses do not map to  A S D F but to X C V B I think

bolob, 2020-12-06:
> With actual master branch ([34bb24a](https://github.com/keyboardio/Kaleidoscope/commit/34bb24a7f1c60bbd393b8b6c1b5480a3bc93d2fb) refs/heads/master), `src/kaleidoscope/plugin/Qukeys.h` line 135:
> 
> ```c++
>   // Set the minimum interval between the previous keypress and the qukey press
>   // to make the qukey eligible to become its alternate keycode.
>   void setMinimumPriorInterval(uint8_t min_interval) {
>     minimum_prior_interval_ = min_interval;
>   }
> ```

 Yep, I found that file. Didn't know that the arduino files were not updated

Rom1deTroyes, 2020-12-06:
What is your keyboard ?

On Model01, asdf is on row2 but for the Atreus, it's on row1...
Does `r0c0` gives you a <kbd>Q</kbd> ?

bolob, 2020-12-06:
Yes, I have an Atreus. This is why then. I hadn't thought about the extra row and column on the Model 01, and it  wasn't clear to me that the .ino code examples for the plugins where common to all the keyboards.

Rom1deTroyes, 2020-12-06:
So if it work as expected with the row/columns shifted to suit the Atreus physical implantation, the issue can be labeled `Qukeys examples has issues with Atreus` and labeled with a `Documentation` tag, so someone write a disambiguation paragraph in the doc :-)

bolob, 2020-12-07:
I modified the title but can't put any label.
How to push the changes of the master in the Arduino ?

jpjerkins, 2021-01-13:
Also interested to know when the Arduino board manager copy will get the latest from master.  Really think the new setMinimumPriorInterval will help with my typing style.

obra, 2021-01-13:
The Boards Manager took an update with the release of v1.99.2 about a week
and a half ago:
https://github.com/keyboardio/Kaleidoscope/releases/tag/v1.99.2

jpjerkins, 2021-01-13:
The problem is that the version number - 1.99.2 - is almost a repeat of the version number from the release three months ago. I say “almost” because the version number from three months ago was prefixed with a “v”, like all the previous versions.

I got the updated Qukeys code to compile locally by hosting my own version of the board index locally, with a new version listed and a corresponding new .tar.bz2 file. But the new Qukeys code won’t be available to most devs until the version number goes over 1.99.2.

Phil Jerkins

obra, 2021-01-13:
Then I screwed up somewhere. Let me go digging.

obra, 2021-01-13:
Crap. I screwed up badly. Apparently there's a bug in the releng tooling
that let me clobber a release with a same-numbered release. Let me see if I
can get this ironed out this evening.

jpjerkins, 2021-01-13:
LOL! I wasn’t trying to keep you up all night! I learned a lot about hosting Arduino boards in the process.

For sure, don’t stay up late for my sake. I’ve already got the feature I wanted into my Atreus :-)

Phil Jerkins

obra, 2021-01-13:
Much appreciated :)  Still a little bit mortifying ;)

gedankenexperimenter, 2021-05-26:
I think it would be okay to close this issue now.

PR #1059 (2021-06-10): Add a few explanatory comments to Qukeys example sketch
This should make it clear that the key coordinates in the `Qukey` example definitions are specific to the Model01, and need to be adjusted for other keyboards.

Closes #991
closed 2021-06-10 by algernon