keyboardio/Kaleidoscope #996: make long-hold more useful

simoninlondon, 2020-12-16:
Chrysalis/Kaleidoscope has "layer shift when held" function. With this function, when a key is long-held, a different layer will be invoked.  I would like to see this function expanded so that when a key is long-held, a new layer is selected AND ALSO the held key on that new layer is triggered.

This will allow auto-shift functions to be implemented, so for instance if a lower case letter is set on the base layer, the upper case variant of the letter could be invoked through a long press.  To this extent, think of this a working much like how a phone keyboard works.  Particularly for the Atreus, this gives access to far more keys/functions without needing multiple key presses.

This advantage this approach offers over a conventional auto-shift approach (which, best I can tell Chrysalis/Kaleidoscope does not offer) is greater control over the "shifted" (long-held) character (since by using a separate layer you're not limited to any key's default shifted option).

And just so we're clear, I'd like to see this as an additional option, not as a replacement for the existing functionality.

As a side issue, if this can be implemented, then it would be nice to have the facility to control how long a key has to be held for a long-press to be registered.

I'm happy to elaborate if anyone wants further explanation.

gedankenexperimenter, 2020-12-16:
The plugin that deals with the "layer shift when held" keys in Chrysalis is called Qukeys, and it can be configured to do the auto-shift that you want with a custom Kaleidoscope sketch. The options available in Chrysalis are very limited, but Qukeys can be configured to use _any_ `Key` value for both the "tap" and "hold" behaviour for a given keymap entry. The way to do it would not be to use a layer shift (which would cause problems with rollover), but instead to put (for example) `Key_A` in the keymap for a given address, and define a Qukey with an alternate (i.e. "hold") value of `LSHIFT(Key_A)`.

See the [Qukeys documentation](https://kaleidoscope.readthedocs.io/en/latest/plugins/Qukeys.html) for more details.

gedankenexperimenter, 2020-12-16:
Alternatively, an auto-shift plugin would not be particularly difficult to write. In fact, there is an existing third-party [AutoShift plugin](https://github.com/tiltowait/Kaleidoscope-AutoShift), but it hasn't been updated in a while.

obra, 2020-12-16:
Thanks for the pointers @gedankenexperimenter! I'd forgotten about AutoShift and the Qukey solution hadn't occurred to me when Simon had asked me previously.

Hrm. Have you given much thought to how we might store QuKeys in eeprom?

gedankenexperimenter, 2020-12-16:
> Have you given much thought to how we might store QuKeys in eeprom?

Only a bit. I keep giving up because of the allocation question: how big should the array be, and how does it get located in EEPROM? Since I've never personally used the EEPROM (building and flashing a new firmware seems easier to me that figuring out Chrysalis), I have only vague notions of how things are done there.

gedankenexperimenter, 2020-12-16:
For this problem, it would be pretty straightforward for me to write a dedicated AutoShift plugin if @tiltowaitt's old one doesn't work with today's Kaleidoscope.