keyboardio/Kaleidoscope #1003: Lift LEDControl to the plugins directory

obra, 2021-01-11:
LEDControl, originally a plugin, has gotten its hooks into the Kaleidoscope core. We should lift it back to the plugins directory, now that that's a thing agian.

algernon, 2021-01-12:
`LEDControl` is referenced from `kaleidoscope/hooks.h`, and that use was added in fddbd75ff89f33a806188340710ca23381458e58, when we implemented the `onLEDModeChange()` hook. I don't really see why we need the forward declaration and the friend class in there, mind you. Hopefully it was an oversight, and in that case, can be safely removed.

I'll run a few tests.

algernon, 2021-01-12:
It's not that easy, afterall... we also use LEDModeManager in core, because we want to generate an array-in-progmem thing for each plugin in `KALEIDOSCOPE_INIT_PLUGINS`, so it _needs_ to be done at the same time as `KALEIDOSCOPE_INIT_PLUGINS`.

What we can do, is move all the files into their proper place, but still use the plugin in core. That doesn't buy us much, except for having LEDControl in its proper place. But it does not remove core's dependency on it.

algernon, 2021-01-12:
Oh, by the way, we _can_ remove the forward declaration and the friend relationship from `kaleidoscope/hooks.h`, but that doesn't buy us much.

obra, 2021-01-12:
Does LEDModeManager need to depend on LEDControl? 

(LEDControl also makes the core depend on Focus)