keyboardio/Kaleidoscope #1004: It should be possible to assign "win-tab" or "alt-tab" to a key in Kaleidoscope or Chrysalis without a custom macro

obra, 2021-01-11:
This is a little bit complicated because the behavior we need is "hold down the modifier, tap tab, release the modifier" 

And to make it really fluid, we likely want to hold down the "alt" in between taps, unless the user presses another key that's not shift.

The right solution -may- be a plugin to add the custom keys.

algernon, 2021-01-12:
The `AppSwitch` example does _something_ similar. We can probably turn that into a plugin, which we can then add to the experimental firmwares in Chrysalis, and expose a special key.

gedankenexperimenter, 2021-06-07:
After the `KeyEvent` handler changes, that AppSwitcher example was converted from using Macros to a small custom plugin. It could certainly be turned into an "official" plugin with very little effort.