keyboardio/Kaleidoscope #1005: Ancient compat stuff is hanging around from before the INIT_PLUGINS system came into being; it should be removed

obra, 2021-01-11:
in `src/kaleidoscope/hooks.cpp`:
```
// The following weak symbols are overwritten by
// using the KALEIDOSCOPE_INIT_PLUGINS(...) macro
// in the the firmware sketch. Their only purpose is
// to provide backwards compatibility during the transition
// from the old implementation of the hooking/plugin system
// to the new one. The weak symbols ensure that there
// are no undefined symbols if KALEIDOSCOPE_INIT_PLUGINS(...)
// is not used. This allows legacy sketches to be used
// during the transition phase.
```

algernon, 2021-01-12:
While this sounds trivial... there's a catch: to be able to remove those weak symbols, we need to have a `KALEIDOSCOPE_INIT_PLUGINS()` invocation in the sketch. Currently, our `examples/Basic` sketch does not have any, because it doesn't use any plugins. Adding an empty one results in a build error, because it requires at least one plugin.

So to be able to remove these compat things, we need to make it possible to use `KALEIDOSCOPE_INIT_PLUGINS()` without arguments.

algernon, 2021-01-12:
Looks like we need `__VA_OPT__` for this, which is available in c++20, supported by gcc8+. Our avr-g++ is older than that, only gcc7. There's https://github.com/willwray/VA_OPT which may help, but it makes things horribly complicated, in my opinion.

I think it'd be more practical to have the legacy symbols in place, and remove them when we upgrade our avr-g++ and can reliably use `__VA_OPT__`.

algernon, 2021-01-12:
Or, as another option: keep the symbols indefinitely, and update the comments to explain why we have them: to support cases where the sketch uses no plugins, thus, no `KALEIDOSCOPE_INIT_PLUGINS()` invocation, because we do not support an empty list in that macro.

algernon, 2021-01-12:
Hmmm.... another option would be to always add a dummy plugin to the list. That's a bit of a hack too, but would allow `KALEIDOSCOPE_INIT_PLUGINS()` to work (hopefully). I still think that plugins that do not use plugins, should be able to get away with not using the macro at all. And for that, we need the weak symbols.

I'll open a PR that updates the comments. If you'd prefer to remove the weak symbols and allow `KALEIDOSCOPE_INIT_PLUGINS()` to work without arguments, I'll explore the dummy plugin option.

noseglasses, 2021-01-12:
> So to be able to remove these compat things, we need to make it possible to use KALEIDOSCOPE_INIT_PLUGINS() without arguments.

You could add a dummy instance of any trivial plugin type, say the plugin base class, to the invocation of `_KALEIDOSCOPE_INIT_PLUGINS` like 
```cpp
#define KALEIDOSCOPE_INIT_PLUGINS(...) \
  kaleidoscope:::Plugin dummyPlugin; \
  _KALEIDOSCOPE_INIT_PLUGINS(dummyPlugin, ##__VA_ARGS__)
```
The compiler should entirely remove the empty hook invocations and you will just have to invest a byte for the instance.

Mind you, this is syntactically untested as my M01 is waiting for a keyswitch repair and I haven't build the firmware for ages :-(

algernon, 2021-01-12:
> You could add a dummy instance of any trivial plugin type, say the plugin base class, to the invocation of `_KALEIDOSCOPE_INIT_PLUGINS` like

I considered that, and it works, indeed, and is correctly optimized out. My dummy plugin was a bit more verbose though, which is why I preferred the current weak symbols. I'll try your solution, and if that works just aswell, I'll PR that instead.

Thanks!

algernon, 2021-01-12:
Ooops, I was wrong - it doesn't work (I had a dummy plugin added in the sketch, too).

Even with an injected dummy, we error out with an empty `KALEIDOSCOPE_INIT_PLUGINS()`, whether I put the injection into `KALEIDOSCOPE_INIT_PLUGINS()` itself or into the `MAP` invocation in `_KALEIDOSCOPE_INIT_PLUGINS()`.

algernon, 2021-01-12:
A'ight, tried using the `VA_OPT` library mentioned above, and while that allows us to have empty `KALEIDOSCOPE_INIT_PLUGINS()`, we get undefined symbols then, if I use the helper macros at the `KALEIDOSCOPE_INIT_PLUGINS()` level. If I use it deeper... well, then the error messages make no sense at all.

noseglasses, 2021-01-12:
> Even with an injected dummy, we error out with an empty KALEIDOSCOPE_INIT_PLUGINS(), whether I put the injection into KALEIDOSCOPE_INIT_PLUGINS() itself or into the MAP invocation in _KALEIDOSCOPE_INIT_PLUGINS().

What do you mean by "injected dummy"?

algernon, 2021-01-12:
```
Building in quiet mode. For a lot more information, add 'VERBOSE=1' to the beginning of your call to make
make -C examples/Basic -f /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/etc/makefiles/sketch.mk compile
make[1]: Entering directory '/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Basic'
In file included from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/Kaleidoscope.h:86:0,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Basic/Basic.ino:18:
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Basic/Basic.ino: In static member function 'static kaleidoscope::EventHandlerResult kaleidoscope_internal::EventDispatcher::apply(Args__&& ...)':
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope_internal/VA_OPT.hpp:141:17: error: 'VA_EAT' was not declared in this scope
 #  define IFN_1 VA_EAT,
                 ^
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/macro_map.h:31:20: note: in definition of macro 'EVAL0'
```

Thus, it seems that even if we used `__VA_OPT__` (or the VA_OPT library), we'd still need the weak symbols. So updating the comments seem to be the best course of action.

algernon, 2021-01-12:
> What do you mean by "injected dummy"?

```c++
#define KALEIDOSCOPE_INIT_PLUGINS(...) \
  kaleidoscope:::Plugin dummyPlugin; \
  _KALEIDOSCOPE_INIT_PLUGINS(dummyPlugin, ##__VA_ARGS__)
```

This. Either here, or at [event_dispatch.h#194](https://github.com/keyboardio/Kaleidoscope/blob/b3b56efdf8bbc1049dfc9ab8e47ae9efdf474c08/src/kaleidoscope_internal/event_dispatch.h#L194). They both fail to compile with the same error is if we didn't use a dummy plugin.

algernon, 2021-01-12:
Hmm.... I have one more idea.

algernon, 2021-01-12:
Looks like I'm succeeding with this:

```c++
#define KALEIDOSCOPE_INIT_PLUGINS(...)                                  \
  kaleidoscope::Plugin __dummy_plugin__;                                \
  _KALEIDOSCOPE_INIT_PLUGINS(                                           \
    IFNE(__VA_ARGS__)(__VA_ARGS__, __dummy_plugin__)                    \
  )
```

Just waiting for the smoke test to finish.

algernon, 2021-01-12:
Looks like I need to fix up sketch exploration too, that's the only thing failing now.

algernon, 2021-01-12:
Not exactly sure _why_ the sketch exploration example is failing, though. It should be called with the same arguments as `_KALEIDOSCOPE_INIT_PLUGINS()`: either with a plugin list, or with a dummy only, yet, the example sketch fails on the static asserts.

algernon, 2021-01-12:
Hm, looks like it gets into the dummy plugin branch.

algernon, 2021-01-12:
Pushed my current state to [hooks/v1-cleanup](https://github.com/keyboardio/Kaleidoscope/tree/hooks/v1-cleanup). Everything compiles fine, except for the sketch exploration example, which fails on asserts:

```
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino: In instantiation of 'kaleidoscope::EventHandlerResult APlugin::exploreSketch() [with _Sketch = kaleidoscope::sketch_exploration::Sketch]':
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:127:1:   required from 'static kaleidoscope::EventHandlerResult kaleidoscope_internal::EventHandler_exploreSketch_v1_caller<hook_is_implemented__, Plugin__, _Sketch, Args__>::call(Plugin__&, Args__&& ...) [with bool hook_is_implemented__ = true; Plugin__ = APlugin; _Sketch = kaleidoscope::sketch_exploration::Sketch; Args__ = {}]'
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:127:1:   required from 'static kaleidoscope::EventHandlerResult kaleidoscope_internal::EventHandler_exploreSketch_v1<_Sketch>::call(Plugin__&, Args__&& ...) [with Plugin__ = APlugin; Args__ = {}; _Sketch = kaleidoscope::sketch_exploration::Sketch]'
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:127:1:   required from 'static kaleidoscope::EventHandlerResult kaleidoscope_internal::EventDispatcher::apply(Args__&& ...) [with EventHandler__ = kaleidoscope_internal::EventHandler_exploreSketch_v1<kaleidoscope::sketch_exploration::Sketch>; Args__ = {}]'
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:127:1:   required from 'static kaleidoscope::EventHandlerResult kaleidoscope::Hooks::exploreSketch() [with _Sketch = kaleidoscope::sketch_exploration::Sketch]'
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:25:1:   required from here
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:96:5: error: static assertion failed
     static_assert(std::is_same<ENTRY_TYPE(P, 1), BPlugin>::value, "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:98:5: error: static assertion failed
     static_assert(P::size == 3, "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:100:5: error: static assertion failed
     static_assert(!ENRTY_IS_LAST(P, 0), "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:101:5: error: static assertion failed
     static_assert(!ENRTY_IS_LAST(P, 1), "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:105:5: error: static assertion failed
     static_assert(PLUGIN_POSITION(P, BPlugin) == 1, "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:109:5: error: static assertion failed
     static_assert(PLUGIN_IS_REGISTERED(P, BPlugin) == true, "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:112:5: error: static assertion failed
     static_assert(NUM_OCCURRENCES(P, APlugin) == 2, "");
     ^~~~~~~~~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/examples/Internal/Sketch_Exploration/Sketch_Exploration.ino:113:5: error: static assertion failed
     static_assert(NUM_OCCURRENCES(P, BPlugin) == 1, "");
     ^~~~~~~~~~~~~
```

Haven't dived deep into it yet, but doing a quick trial & error, changing `static_assert(P::size == 3, "");` to `P::size == 1` made it pass, which is super weird. All the errors seem to suggest we're on the dummy plugin branch, while we really should not be.

algernon, 2021-01-12:
Aight. Managed to fix that, I think. Moved the emptyness checks elsewhere.

commit 94b1cd30 (2021-01-12): kaleidoscope/hooks.cpp: Update a comment to reflect reality

The `INSTANTIATE_WEAK_HOOK_FUNCTION` claimed that it's there for v1 API
compatibility alone - it is not. It allows us to have sketches that use no
plugins too, without them having to use `KALEIDOSCOPE_INIT_PLUGINS()` with a
dummy plugin.

We'd need a dummy plugin because `KALEIDOSCOPE_INIT_PLUGINS()` does not support
being invoked with an empty plugin list, due to technical reasons. From an
end-user point of view, not using the macro is much preferable to using it with
a dummy plugin. We can't automatically inject a dummy plugin eith [... cut]
commit 7f8f9d68 (2021-01-12): hooks: Remove V1 backwards compatibility

Now that we can have an empty `KALEIDOSCOPE_INIT_PLUGINS()` invocation, we can
safely remove the V1 plugin API backwards compatibility code and weak symbols
from the hooks system.

Together with the previous commits, this fixes #1005.
commit c924fb0d (2021-01-12): Update a comment to reflect reality

The `INSTANTIATE_WEAK_HOOK_FUNCTION` claimed that it's there for v1 API
compatibility alone - it is not. It allows us to have sketches that use no
plugins too, without them having to use `KALEIDOSCOPE_INIT_PLUGINS()` with a
dummy plugin.

We'd need a dummy plugin because `KALEIDOSCOPE_INIT_PLUGINS()` does not support
being invoked with an empty plugin list, due to technical reasons. From an
end-user point of view, not using the macro is much preferable to using it with
a dummy plugin. We can't automatically inject a dummy plugin either, again, due
to techni [... cut]
closed 2021-01-12 by obra