keyboardio/Kaleidoscope #1009: MoveToLayer should cancel layer shifts to the same layer

algernon, 2021-01-25:
This is based on keyboardio/Chrysalis#620. 

The use case is having a DualUse key shift to a particular layer, which has a `MoveToLayer()` key that activates the same layer. The expected behaviour would be that once the `MoveToLayer()` key is hit, releasing the dual use shift would keep us at the layer we moved to, rather than deactivating the layer, and moving back to the default.

What happens under the hood is that when the dual use shift is held, we shift to that layer. When `MoveToLayer` is activated next, we clear every layer, and (re-)activate the given one. Then, when we release the layer shift key, we end up at the default layer, because we released after activating `MoveToLayer`. I believe this is not intuitive behaviour.

When we use `MoveToLayer`, we want to move to that layer, and remain there. As such, we should turn any active layer shift keys (be those normal layer shifts, dual use shifts, or oneshot shifts) that target the same layer as the `MoveToLayer` targets to no-ops. Thus, when they're released, we do not deactivate the moved to layer. Since `MoveToLayer` deactivates all other layers, we can simplify this to turning all active layer shifts into no-ops.

This would be a behaviour change, and I believe it should only be done for `MoveToLayer`, not for `LockToLayer`, but it would result in more intuitive behaviour.

gedankenexperimenter, 2021-01-26:
In the case where both target layers are the same, I think it should also apply to `LockLayer`, not just `MoveToLayer`. On a basic keyboard, if I press  <kbd>shift</kbd>, press <kbd>CapsLock</kbd>, then release <kbd>shift</kbd>, I'll still get capital letters when I type. It would be rather surprising to have `ShiftToLayer` and `LockLayer` have a different behaviour.

On another note, the implementation of this would be trivial with #904 merged—just one pass through the array, changing any `ShiftToLayer` keys to `Key_NoKey` for `MoveToLayer` toggle on, and any `ShiftToLayer(N)` keys to `Key_NoKey` for `LockLayer(N)` key toggle on.

necsoiu, 2021-02-03:
I think we need to consider blocking the layer rather than moving on to it.
In other words when we are shifted to a layer we must to have a way to block that layer pressing one key  (e.g. lock to layer key). Pressing again the same key must unlock the shifted layer.
I think this is an absolutely necessary feature on a keyboard with so few keys, otherwise we will have to use(block) a key on level 0 which we will use to lock and unlock a certain level.
In fact I think the title of this bug should be '**Locking/unlocking the active shifted layer**'.

gedankenexperimenter, 2021-02-03:
I've thought about this a bit more, and there's more complexity here than I first thought for the `LockLayer` case. Consider the following sequence:

1. `LockLayer(X)` tap
2. `ShiftToLayer(X)` down
3. `LockLayer(X)` tap

To emulate <kbd>shift</kbd> & <kbd>CapsLock</kbd>, step 3 above should be a no-op. But without step 1, it should lock layer `X` as active. We don't currently have a structure that allows us to record whether or not step 1 occurred, so it's not really clear what to do.

Also consider this sequence:

1. `LockLayer(X)` tap
2. `ShiftToLayer(X)` down
3. `ShiftToLayer(X)` up

Again, to make the behaviour as similar as possible to <kbd>shift</kbd> & <kbd>CapsLock</kbd>, to abide by the principle of least surprise, we would need to track both the "locked" state of layer `X` _and_ the "shifted" state of layer `X`.

I realize that this is technically orthogonal to the original issue (because `MoveToLayer` is not a toggle, it doesn't incur the same complexity), but it is closely related, and both should probably be addressed together.

gedankenexperimenter, 2021-05-18:
I've been playing around with alternative layer-change semantics for the past few days, including a fix for the layer shift cancelling a layer move problem. I think it provides a more consistent, less surprising set of behaviours.

I've got a working version that stores different values for layer shifts and layer locks on the stack, allowing for both a layer to appear on the active layers stack twice (once shifted, once locked), in different places. There are some cases where it would be reasonable to disagree about how to abide by the principle of least surprise, of course.

The way it works is that layer lock keys always bring the target layer to the top of the layer stack, unless that layer is already there (the argument being that pressing such a key should have a user-visible change). Layer shift keys put the shifted version of the target layer on the top of the stack when toggled on, but only if the shifted version of that layer isn't already on the stack (the idea is that if I'm holding a layer shift key, I would not expect anything to change if I press a second layer shift key for the same target layer). When toggled off, layer shift keys remove the shifted version of the target layer from the stack, but only if there's no other layer shift key for the same layer active. Any locked version of that layer is unaffected, so locking layer `N`, then pressing and releasing `ShiftToLayer(N)` does not deactivate layer `N`.

Layer move keys would be unaffected, except that layer shift keys would no longer deactivate the resultant locked layer.

The two special "next" and "previous" layer shift keys get changed such that they bring the shifted version of the target layer (the current top layer ± 1) to the top of the layer stack, then become a normal `ShiftToLayer()` key for that layer, so that when released, they simply deactivate the layer shift that they caused, rather than deactivating whatever layer is now on top of the stack.

All this is on top of #1024, of course.