keyboardio/Kaleidoscope #1010: Macro keys appear to be broken

algernon, 2021-01-26:
**Describe the bug**

Reported as part of keyboardio/Chrysalis#630, macro keys no longer seem to function. Having a key mapped to macro, pressing the key no longer triggers the macro. Downgrading to an earlier firmware shipped with Chrysalis restores the functionality.

**To Reproduce**

We're assuming here we're using a sketch correctly set up, with the Macros plugin enabled, and a macro implemented. For example, the default Atreus sketch is one such.

1. Flash a Keyboardio Atreus with the latest firmware shipped with Chrysalis, or a firmware built from latest Kaleidoscope sources.
2. Map a key to `Macro #1`, and save changes.
3. Press the key.
4. Observe the macro is not triggered.

**Expected behavior**

The macro should still run, and print `Keyboardio Atreus - Kaleidoscope locally built` (or something similar).

**Additional context**

I have bisected Kaleidoscope between 61e3913f (used for Chrysalis 0.7.8 and 0.7.9 builds, where Macros worked with the shippe firmware) and master, and ended up on ec7cdbcecbb4fc7fd3d30e26d0dc75227ee22710 being the culprit. Reverting that commit on master also fixes the problem.

algernon, 2021-01-26:
Turns out, there's a very simple reason behind this failing: we changed the macro keycodes. Whereas previously Macro #0 was `24576`, `ranges::MACRO_FIRST` is `53596`. Thus the migration to ranges was not backwards compatible.

I'll have a fix coming up soon.

closed 2021-01-26 by algernon

gedankenexperimenter, 2021-01-26:
I guess I failed to consider Chrysalis keycode values when standardizing the range for Macros. However, restoring the Macros `Key` range to its old values is also not a great solution. It means that some simple tests need to become more complex ("Is this a plugin-specific key?" & "Is this a `SYNTHETIC` key?"), and it's more likely that bugs will be triggered & introduced. The change to the Macros key range was not just about symbolically referring to something defined in the `ranges` enum; it was primarily about properly separating the values used for built-in keycodes from those specific to plugins, and your fix undoes that improvement to Kaleidoscope.

We should open PRs for both Kaleidoscope and Chrysalis so that this can be (re-)standardized properly when you decide to give Chrysalis a major version increase (whether that's because of this or some other backwards-incompatible change).

gedankenexperimenter, 2021-01-27:
It's not so much that we need a _better_ solution; it's that we shouldn't squander the next opportunity to move forward. The idea behind opening an issue for Chrysalis in addition to the PR for Kaleidoscope is to serve as a reminder, since fixing this in Kaleidoscope is dependent on improvements to Chrysalis, not _vice versa_ (otherwise, you wouldn't have reverted the standardization of Macros keycodes).

commit 9c49d5b2 (2021-01-26): plugins/Ranges: Fix the Macro key ranges

Macro keycodes pre-date the Ranges plugin, so they previously had a keycode that
sorted before ranges::FIRST. For the sake of backwards compatibility, we want to
keep using the same keycodes.

Fixes #1010.