keyboardio/Kaleidoscope #1011: Macros write in QWERTY on my AZERTY system

Rom1deTroyes, 2021-01-26:
**Describe the bug**
If the OS keyboard is not in QWERTY, the Macros are send as if you use a qwerty layout. 

**To Reproduce**
Steps to reproduce the behavior:

- Change your OS layout to AZERTY
- Press the Macro#1 key
- See the keyboard write `Keyboqrdio Qtreus ) Kqleidoscope locqlly built`

**Expected behavior**
I expect macros to write the strings I've stored in the keyboard

**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Version : Debian10
 - Device Keyboardio Model01 and Keyboardio Atreus

gedankenexperimenter, 2021-01-27:
This is probably not fixable. I'm pretty sure there's no way for the keyboard to know what the host's keymap is set to, and as it sends keycodes, not characters, Macros would need a separate lookup table for each supported OS keymap, which would very quickly exhaust the available storage on the AVR chips used in most keyboards that run Kaleidoscope.

The existing hard-coded QWERTY mapping takes advantage of the fact that the some groups of keycodes (letters especially) are in the same order in the HID report and ASCII, and so does not need a full lookup table. That's a unique advantage of QWERTY, and a good reason to set the OS keymap to QWERTY, and use the programmability of Kaleidoscope to set the keymap on the keyboard instead.

algernon, 2021-04-05:
This is, indeed, not easily fixable. We _could_ make it possible to have multiple lookup tables and select one at compile time, so that we don't run out of space, but that'd still have the problem of working with a single OS layout only. I also think it'd be a bit too much effort to have those tables within Kaleidoscope.

For the time being, I'd rather document this limitation of `Macros.type()` (and Macros in general, as keycodes assume QWERTY).

obra, 2021-04-05:
Moving to 'recorded' macros set up in Chrysalis may give us the rope we
need to make this work right for users, since Chrysalis will be able to
display the "right" characters and record keycodes.

algernon, 2021-04-05:
Indeed, EEPROM-based macros do not necessarily suffer from this problem, and that's another way to address the issue - but comes with its own limitations too.