keyboardio/Kaleidoscope #1015: Question about the delay in `readCols()`

nothien, 2021-02-16:
I'm writing custom firmware for my Atreus, and in trying to understand how everything works at the lowest level, I came across the `readCols()` function (`src/kaleidoscope/driver/keyscanner/ATmega.h`).  The comments above it explicitly state that the hardware needs enough time to produce stable reads, and so loop unrolling is disabled.  I then took a look at the produced (dis)assembly:

```
$ avr-objdump -dC atreus.ino.elf
00000736 <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]>:
     736:	ee e3       	ldi	r30, 0x3E	; 62
     738:	f1 e0       	ldi	r31, 0x01	; 1
     73a:	50 e0       	ldi	r21, 0x00	; 0
     73c:	40 e0       	ldi	r20, 0x00	; 0
     73e:	70 e0       	ldi	r23, 0x00	; 0
     740:	60 e0       	ldi	r22, 0x00	; 0
     742:	00 00       	nop
     744:	91 91       	ld	r25, Z+
     746:	a9 2f       	mov	r26, r25
     748:	b0 e0       	ldi	r27, 0x00	; 0
     74a:	84 e0       	ldi	r24, 0x04	; 4
     74c:	b5 95       	asr	r27
     74e:	a7 95       	ror	r26
     750:	8a 95       	dec	r24
     752:	e1 f7       	brne	.-8      	; 0x74c <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0x16>
     754:	90 96       	adiw	r26, 0x20	; 32
     756:	2c 91       	ld	r18, X
     758:	30 e0       	ldi	r19, 0x00	; 0
     75a:	9f 70       	andi	r25, 0x0F	; 15
     75c:	02 c0       	rjmp	.+4      	; 0x762 <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0x2c>
     75e:	35 95       	asr	r19
     760:	27 95       	ror	r18
     762:	9a 95       	dec	r25
     764:	e2 f7       	brpl	.-8      	; 0x75e <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0x28>
     766:	20 95       	com	r18
     768:	30 95       	com	r19
     76a:	21 70       	andi	r18, 0x01	; 1
     76c:	33 27       	eor	r19, r19
     76e:	04 2e       	mov	r0, r20
     770:	02 c0       	rjmp	.+4      	; 0x776 <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0x40>
     772:	22 0f       	add	r18, r18
     774:	33 1f       	adc	r19, r19
     776:	0a 94       	dec	r0
     778:	e2 f7       	brpl	.-8      	; 0x772 <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0x3c>
     77a:	62 2b       	or	r22, r18
     77c:	73 2b       	or	r23, r19
     77e:	4f 5f       	subi	r20, 0xFF	; 255
     780:	5f 4f       	sbci	r21, 0xFF	; 255
     782:	4c 30       	cpi	r20, 0x0C	; 12
     784:	51 05       	cpc	r21, r1
     786:	e9 f6       	brne	.-70     	; 0x742 <kaleidoscope::driver::keyscanner::ATmega<kaleidoscope::device::keyboardio::AtreusProps::KeyScannerProps>::readCols() [clone .isra.26]+0xc>
     788:	cb 01       	movw	r24, r22
     78a:	08 95       	ret
```

The code essentially iterates over different columns, reads the pin info, right-shifts (via a loop) the info 4 times to get the port address, reads all 8 pins in the port, right-shifts (via a loop) to the correct pin number, and then left-shifts (via a loop) to the correct column number.

I looked up the manual for the ATmega32U4 [here](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf) and calculated the total cycle count between individual reads.  It comes up to about 87, which is unexpectedly large, given that it can be implemented with only 2-3 cycles per pin using `sbic` - `sbr` pairs.  There's a lot of slowdown, particularly for bitshifts, because they have to be looped.  That also means that the NOP inserted in `readCols()` has a very small effect in the delay.

I have a few questions:
1. Do you have any experimental data on minimum / maximum acceptable delays between reads?  If not, can you suggest any way to calculate this?  `readCols()` is probably the hottest function around, and so I do want to make it as fast as it can be, particularly when it seems to be unnecessarily slow.
2. The comments mention that the hardware needs the delays in order to stabilize reads, but it reads the entire port (all 8 pins) every loop, and in many cases multiple pins connected to the keys are on the same port.  Is the hardware not reading all 8 pins correctly?  If it is reading them correctly, couldn't the work be safely reduced by reading only once from each port?  This would require special casing, which would be arduous, but it would work well in my case because I only need to support the Atreus.
3. Where can I find more information about what's going on at the hardware level here?  The comments mention 'hot pins' and I assume that the hardware has to load the current column of keys after the row selection which can take time, but I don't yet have a good understanding of how long that can take and what other factors (if any) have to be considered.

obra, 2021-02-16:
Experimentally, we've seen issues. But I'd run with the datasheet's
instructions

nothien, 2021-02-17:
Alright, thanks!  I'll do some testing on my own board and see if it has an effect.

algernon, 2021-04-05:
> Do you have any experimental data on minimum / maximum acceptable delays between reads? If not, can you suggest any way to calculate this? readCols() is probably the hottest function around, and so I do want to make it as fast as it can be, particularly when it seems to be unnecessarily slow.

While I don't have exact numbers, my experience has been that the current amount of delays and slowdowns result in reliable scanning. Enabling loop unrolling however, resulted in erratic behaviour: missed keys, chatter, that kind of thing. With unrolling enabled, those were _very_ easy to reproduce with just normal use of the keyboard.

I have not made attempts at measuring further, because the status quo is good enough for the most part.

If you end up with a faster scanner, that works reliable, I'd be thrilled to have a look at it!

closed 2021-09-22 by obra