keyboardio/Kaleidoscope #1020: Add next/previous track keys

Vlek, 2021-03-22:
**Is your feature request related to a problem? Please describe.**
I am currently unable to map keys to perform the media functions "play next track" and "play previous track"

**Describe the solution you'd like**
I would like to be able to have an easy-to-use key, named something like Key_MediaTrackNext and Key_MediaTrackPrevious to use in key mapping.

**Describe alternatives you've considered**
Currently available keys may offer some similar functionality, but I am not knowledgeable enough in the different key events to know which would be able to function in the same way as the ones I am requesting.

**Additional context**
None.

gedankenexperimenter, 2021-03-22:
Have you tried `Consumer_ScanNextTrack` and `Consumer_ScanPreviousTrack`?

Vlek, 2021-03-22:
That did it for me! Thank you.

closed 2021-03-22 by Vlek