keyboardio/Kaleidoscope #1022: Errors when running `make docker-clean`

gedankenexperimenter, 2021-03-27:
**Describe the bug**
When running `make docker-clean` on macOS, I'm seeing an error message for each testcase directory:
```
Error creating instance: error loading platform index:
loading json index file /arduino-cli/data/package_keyboardio_index.json: open /arduino-cli/data/package_keyboardio_index.json: no such file or directory
```
This doesn't appear to cause any real problems.

**To Reproduce**
```
$ make docker-clean
```

**Expected behavior**
No error messages.

**Environment (please complete the following information):**
 - OS: macOS
 - Version: 11.2.3
 - Device: MacBook Air (M1, 2020)

**Additional context**
I haven't verified this, but it looks like it's no longer possible to do a `docker-clean`, then run `docker-simulator-tests` without a functional internet connection. It would be nice to be able to run tests while offline, even if `make setup` can only be run while connected. Sometimes code changes result in linker errors that can only be resolved by running `make docker-clean`, and it looks like the package index file needs to be re-downloaded after running it.

obra, 2021-03-28:
@gedankenexperimenter does #1023 fix this for you?

gedankenexperimenter, 2021-03-28:
This does fix the error messages. It doesn't address the other item, with Docker 3.3.0 on M1, but that's a separate thing, so I'm closing this issue.

closed 2021-03-28 by gedankenexperimenter