keyboardio/Kaleidoscope #1031: In Windows, Unicode.type does not produce a Unicode character

fpsvogel, 2021-04-20:
**Describe the bug**
Instead of producing the desired Unicode character, `Unicode.type(character)` produces the hex code or (if `HostOS.os` is set) unexpected key combinations.

**To Reproduce**

Below is the Unicode-related parts of my sketch. [Here is the entire sketch.](https://pastebin.com/24i9HGj3)

This code follows [the official guide](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-Unicode.html) and [example](https://kaleidoscope.readthedocs.io/en/latest/examples/Keystrokes/Unicode/Unicode.ino.html). When I adapt the example and use it as my Atreus layout, pressing the keyboard-symbol key produces "2328 " instead of “:keyboard:”. This is the same kind of result that I get from the code below.

```
#include "Kaleidoscope-HostOS.h"
#include "Kaleidoscope-Unicode.h"

// ...

KALEIDOSCOPE_INIT_PLUGINS(
  // ...
  HostOS,
  Unicode
);

static void unicode(uint32_t character, uint8_t keyState) {
  if (keyToggledOn(keyState)) {
    Unicode.type(character);
  }
}

const macro_t *macroAction(uint8_t macroIndex, uint8_t keyState) {
  switch (macroIndex) {
  // ...
  case EM_DASH:
    unicode(0x2014, keyState);
    break;
  case EN_DASH:
    unicode(0x2013, keyState);
    break;
  case LEFT_SINGLE_QUOTE:
    unicode(0x2018, keyState);
    break;
  case RIGHT_SINGLE_QUOTE:
    unicode(0x2019, keyState);
    break;
  // other characters omitted ...
  }

  return MACRO_NONE;
}

// ...
```
When I type the keys for these four characters in sequence, the following text is produced (including a space after each number):

`2014 2013 2018 2019 `

I also tried adding `HostOS.os(kaleidoscope::hostos::WINDOWS);` to the end of my `setup()`, but then unexpected key combinations are produced. These combinations include undisplayable characters and/or modifier keys (in some cases tabs are switched or menu options are selected in the active window).

**Expected behavior**
`Unicode.type(character)` should produce the Unicode character specified by the hex code in `character`.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Version 2004, build 19041.928
 - Device: Keyboardio Atreus

algernon, 2021-06-16:
It appears our documentation is missing a crucial detail: for Unicode input to work on Windows, you'll have to edit a registry setting: under `HKEY_Current_User/Control Panel/Input Method`, set `EnableHexNumpad` to `"1"`. If such a key does not exist, you need to create it, and set its type to `REG_SZ`. Once that's done, unicode input should work.

I'll update our documentation to reflect this.

closed 2021-06-16 by obra

fpsvogel, 2021-06-18:
@algernon Thanks for your reply. However, I am still getting the same results. It is not clear to me whether I should include quote marks in the value for `EnableHexNumpad` (as in your reply), or whether I should include the line `HostOS.os(kaleidoscope::hostos::WINDOWS);` in my sketch, so I tried all four possibilities (with/without quote marks, and with/without the `HostOS.os` line). Each time I made a registry change, I restarted my computer to make sure that it was applied.

But I got the same results as before: hex codes if omitting the `HostOS.os` line, or seemingly random characters and modifiers if including the `HostOS.os` line.

fpsvogel, 2021-09-02:
To give this thread closure: I gave it another try and got the same results, but this time I saw that Kaleidoscope's Unicode input is fine, but my trouble with it is caused by two external factors:

* Unicode input in Windows relies on` Alt + [a hex code]`, which may activate shortcuts involving `Alt` (which is especially likely when the hex code includes letters). [Here's a discussion on this](https://superuser.com/questions/864944/cant-type-unicode-because-of-keyboard-shortcuts). In short, there's no solution except to disable all `Alt` shortcuts, which I don't want to do because I use these shortcuts.
* My Kaleidoscope layout is QWERTY but I use a Colemak Windows keyboard layout, which means that the hex code sent by the keyboard sometimes ends up garbled. For example, "201D" becomes "201S" because in Colemak, the "S" key is in the same position as the QWERTY "D" key.

So Unicode input via Kaleidoscope just won't work, because of my particular situation (I like my `Alt` shortcuts, and I use a non-QWERTY Windows keyboard layout).

In the end I set up the Unicode characters in AutoHotkey, and that will work well.

PR #1067 (2021-06-16): Unicode: Update the README with Windows instructions
Under Windows, we use an input method that requires a registry edit, mention that in the README, and describe how to do it.

Fixes #1031.
commit d34aef14 (2021-06-16): Unicode: Update the README with Windows instructions

Under Windows, we use an input method that requires a registry edit, mention
that in the README, and describe how to do it.

Fixes #1031.