keyboardio/Kaleidoscope #1061: Custom modifiers for screen readers

Jookia, 2021-06-13:
**Is your feature request related to a problem? Please describe.**
Currently I can one-shot press modifiers, or sticky hold them, but I can't do this with regular keys. Some applications have their own fake modifiers that I'd like to use.

**Describe the solution you'd like**
I'd like to be able to add custom modifiers, like insert and caps lock.

**Describe alternatives you've considered**
Holding keys down kinda sucks.

**Additional context**
See here for some common screen reader keybinds:
https://help.gnome.org/users/orca/stable/preferences_key_bindings.html.en

gedankenexperimenter, 2021-06-13:
OneShot has two very new features that might be good enough, accessible by putting one of two special OneShot keys in your keymap:
- `OneShot_MetaStickyKey` — tap this, then the next key you tap will become "sticky" until the next time it gets pressed.
- `OneShot_ActiveStickyKey` — hold any key(s), then tap this key, and all the held ("active") keys will become "sticky".

Jookia, 2021-06-13:
Those do look helpful, but aren't exactly what I'd like. Is there a
function call I could make to trigger stickiness for a certain key?

gedankenexperimenter, 2021-06-13:
It would also be possible to modify OneShot or (I'm pretty sure, but I'll have to check) add a simple custom plugin to a sketch that will turn any key you define into a OneShot key; that might be the best solution.

We could consider defining the auto-OneShot detection function weakly, so it can be overridden in user sketches, as well.

gedankenexperimenter, 2021-06-13:
I've just had a look, and OneShot doesn't currently expose a way to make keys sticky, but I think it definitely should. I'm hopeful that will even let us move the handling of the aforementioned special keys into a separate plugin.

algernon, 2021-06-13:
The major reason OneShot is restricted to layer and modifier keys is quite simple: space. OneShot stores the active OneShot keys in a 16-bit bitmap, 8 modifiers + 8 layer keys. If we increase that to 24 or 32 bits, the code size and memory usage will grow prohibitively much.

What we _could_ do, is allow one to set what keys are considered modifiers. That would still limit us to 8 modifiers total (or 16 at max, if we make it possible to have 0 OSL keys), but perhaps that's enough for the time being.

alfalfasprossen, 2021-06-13:
I might have a plugin that does what you need https://github.com/alfalfasprossen/Kaleidoscope-Repeater
I don't know if it compiles against the current state of Kaleidoscope though, so let me know - I would like to update all my stuff to the latest Kaleidoscope anyway one day 😁

gedankenexperimenter, 2021-06-13:
> The major reason OneShot is restricted to layer and modifier keys is quite simple: space. OneShot used to store the active OneShot keys in a 16-bit bitmap, 8 modifiers + 8 layer keys. It doesn't do that anymore, but stickable keys are still stored in a 16-bit bitmap.

Good point. That applies to keys that are in the "one-shot" state, for when they get double-tapped, but not keys that get put directly in the "sticky" state. We should probably make the `isStickable(key)` function override-able, too.

gedankenexperimenter, 2021-06-13:
> I might have a plugin that does what you need https://github.com/alfalfasprossen/Kaleidoscope-Repeater
> I don't know if it compiles against the current state of Kaleidoscope though, so let me know - I would like to update all my stuff to the latest Kaleidoscope anyway one day 😁 

It probably _compiles_, but it's unlikely that it will work as expected with the current Kaleidoscope master branch. I recently started writing a "plugin author's guide" that might be useful to you. When you get around to updating any plugins, I'd love to hear any feedback you have on how to improve it.

gedankenexperimenter, 2021-06-14:
With #1062, it is possible to add the following custom plugin to a sketch, which will make the `insert` key into a OneShot key:

```c++
namespace kaleidoscope {
namespace plugin {

class OneShotInsert : public Plugin {
 public:
  EventHandlerResult onKeyEvent(KeyEvent &event) {
    if (keyToggledOn(event.state) && (event.state & INJECTED) == 0 &&
        event.key == Key_Insert && live_keys[event.addr] != event.key)
      ::OneShot.setPending(event.addr);
    return EventHandlerResult::OK;
  }
};

} // namespace plugin
} // namespace kaleidoscope

kaleidoscope::plugin::OneShotInsert OneShotInsert;

KALEIDOSCOPE_INIT_PLUGINS(OneShot, OneShotInsert);
```

Jookia, 2021-06-15:
I'll test this when I get a chance for my use cases, but that's pretty cool!