keyboardio/Kaleidoscope #1064: Unicode plugin: bad behaviour on Linux when multiple input methods are available

BenjaminGalliot, 2021-06-15:
Currently on Linux, when the system is configured with one simple input type (French, English...), everything works fine. However, if another input system, indirect like Chinese (pinyin) or Japanese (romaji), is added, even if it is not the active input, the Unicode input is blocked (only the `u` character is typed, in general).

The only way I found to solve the problem is to force the use of the numeric keypad digits on Linux, as on Windows.

For the code, it consisted in replacing
```
key = hexToKey(digit);
```
by
```
key = hexToKeysWithNumpad(digit);
```

I didn't understand the origin of the interaction between the software part (several input sources) and the difference in treatment between the numeric keypad and the numeric top bar (on this bar, digits are on shift layer in French, by the way, unlike in English).

I don't know if the use of the numeric keypad should be made general in this case, especially because I don't know OSX.

algernon, 2021-06-16:
There appears to be no problem with using the numpad on Linux, so I'm inclined to make that the default there too, and leave the number row for macOS only.

algernon, 2021-06-16:
Hmm... one concern I have is that if we use the numpad, we'll depend on numlock state. But then, we already do that on Windows, so it shouldn't be a big concern - and if it turns out to be one, we can always figure out something _then_.

PR #1066 (2021-06-16): Unicode: Use the numpad to input digits on Linux too
When there are different input methods available on Linux (Chinese, Japanese, etc), using the numpad is more reliable than the number row. As both work, rather than making it configurable, just switch to numpad on Linux.

Fixes #1064.
commit ec3acc3d (2021-06-16): Unicode: Use the numpad to input digits on Linux too

When there are different input methods available on Linux (Chinese, Japanese,
etc), using the numpad is more reliable than the number row. As both work,
rather than making it configurable, just switch to numpad on Linux.

Fixes #1064.
closed 2021-06-16 by obra