keyboardio/Kaleidoscope #1074: Issues found attempting to use AutoShift (with Chrysalis)

joe-sharp, 2021-07-03:
**Describe the bug**
First, let me say I am new to Kaleidoscope but have a decent head start having worked with Arduino quite a bit in the past. I heard that there was a brand new plugin added to Kaleidoscope that would allow me to enable AutoShift similar to another keyboard I use. My Atreus is also new so I liked the idea of still being able to configure my keymaps in Chrysalis so I was attempting to add the needed lines of code to firmware already setup to work with Chrysalis.

Issues encountered:
- AutoShift functionality doesn't get enabled
- I start encountering weird ghosting issues issues if I try typing multiple keys quickly. For example quickly pressing "arst" (I am on colemak so "asdf" for QWERTY users) I end up getting "arrrrrrrrrrrrrrrrrrrrrr" and the "r" will keep repeating until I press something else.
- The instructions on the AutoShift plugin page (https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-AutoShift.html) does not mention `AutoShiftConfig` but the example usage has it intialized with the `AutoShift` plugin.
- The instructions describe the enable/disable calls backwards.
```
To turn the plugin off, call AutoShift.enable().

To turn the plugin on, call AutoShift.disable().
```

**To Reproduce**
Attempted to flash the following firmware:
<details><summary>CustomAtreus.ino</summary>

```
/* -*- mode: c++ -*-
 * Atreus -- Chrysalis-enabled Sketch for the Keyboardio Atreus
 * Copyright (C) 2018, 2019  Keyboard.io, Inc
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License along
 * with this program; if not, write to the Free Software Foundation, Inc.,
 * 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
 */

#ifndef BUILD_INFORMATION
#define BUILD_INFORMATION "locally built"
#endif

#include "Kaleidoscope.h"
#include "Kaleidoscope-AutoShift.h"
#include "Kaleidoscope-EEPROM-Settings.h"
#include "Kaleidoscope-EEPROM-Keymap.h"
#include "Kaleidoscope-FocusSerial.h"
#include "Kaleidoscope-Macros.h"
#include "Kaleidoscope-MouseKeys.h"
#include "Kaleidoscope-OneShot.h"
#include "Kaleidoscope-Qukeys.h"
#include "Kaleidoscope-SpaceCadet.h"

#define MO(n) ShiftToLayer(n)
#define TG(n) LockLayer(n)

enum {
  MACRO_QWERTY,
  MACRO_VERSION_INFO
};

#define Key_Exclamation LSHIFT(Key_1)
#define Key_At LSHIFT(Key_2)
#define Key_Hash LSHIFT(Key_3)
#define Key_Dollar LSHIFT(Key_4)
#define Key_Percent LSHIFT(Key_5)
#define Key_Caret LSHIFT(Key_6)
#define Key_And LSHIFT(Key_7)
#define Key_Star LSHIFT(Key_8)
#define Key_Plus LSHIFT(Key_Equals)

enum {
  QWERTY,
  FUN,
  UPPER
};

/* *INDENT-OFF* */
KEYMAPS(
  [QWERTY] = KEYMAP_STACKED
  (
       Key_Q   ,Key_W   ,Key_E       ,Key_R         ,Key_T
      ,Key_A   ,Key_S   ,Key_D       ,Key_F         ,Key_G
      ,Key_Z   ,Key_X   ,Key_C       ,Key_V         ,Key_B, Key_Backtick
      ,Key_Esc ,Key_Tab ,Key_LeftGui ,Key_LeftShift ,Key_Backspace ,Key_LeftControl

                     ,Key_Y     ,Key_U      ,Key_I     ,Key_O      ,Key_P
                     ,Key_H     ,Key_J      ,Key_K     ,Key_L      ,Key_Semicolon
       ,Key_Backslash,Key_N     ,Key_M      ,Key_Comma ,Key_Period ,Key_Slash
       ,Key_LeftAlt  ,Key_Space ,MO(FUN)    ,Key_Minus ,Key_Quote  ,Key_Enter
  ),

  [FUN] = KEYMAP_STACKED
  (
       Key_Exclamation ,Key_At           ,Key_UpArrow   ,Key_Dollar           ,Key_Percent
      ,Key_LeftParen   ,Key_LeftArrow    ,Key_DownArrow ,Key_RightArrow       ,Key_RightParen
      ,Key_LeftBracket ,Key_RightBracket ,Key_Hash      ,Key_LeftCurlyBracket ,Key_RightCurlyBracket ,Key_Caret
      ,TG(UPPER)       ,Key_Insert       ,Key_LeftGui   ,Key_LeftShift        ,Key_Delete         ,Key_LeftControl

                   ,Key_PageUp   ,Key_7 ,Key_8      ,Key_9 ,Key_Backspace
                   ,Key_PageDown ,Key_4 ,Key_5      ,Key_6 ,___
      ,Key_And     ,Key_Star     ,Key_1 ,Key_2      ,Key_3 ,Key_Plus
      ,Key_LeftAlt ,Key_Space    ,___   ,Key_Period ,Key_0 ,Key_Equals
   ),

  [UPPER] = KEYMAP_STACKED
  (
       Key_Insert            ,Key_Home                 ,Key_UpArrow   ,Key_End        ,Key_PageUp
      ,Key_Delete            ,Key_LeftArrow            ,Key_DownArrow ,Key_RightArrow ,Key_PageDown
      ,M(MACRO_VERSION_INFO) ,Consumer_VolumeIncrement ,XXX           ,XXX            ,___ ,___
      ,MoveToLayer(QWERTY)   ,Consumer_VolumeDecrement ,___           ,___            ,___ ,___

                ,Key_UpArrow   ,Key_F7              ,Key_F8          ,Key_F9         ,Key_F10
                ,Key_DownArrow ,Key_F4              ,Key_F5          ,Key_F6         ,Key_F11
      ,___      ,XXX           ,Key_F1              ,Key_F2          ,Key_F3         ,Key_F12
      ,___      ,___           ,MoveToLayer(QWERTY) ,Key_PrintScreen ,Key_ScrollLock ,Consumer_PlaySlashPause
   )
)
/* *INDENT-ON* */

KALEIDOSCOPE_INIT_PLUGINS(
  EEPROMSettings,
  EEPROMKeymap,
  Focus,
  FocusEEPROMCommand,
  FocusSettingsCommand,
  AutoShift,
  Qukeys,
  SpaceCadet,
  OneShot,
  Macros,
  MouseKeys
);

const macro_t *macroAction(uint8_t macro_id, KeyEvent &event) {
  switch (macro_id) {
  case MACRO_QWERTY:
    // This macro is currently unused, but is kept around for compatibility
    // reasons. We used to use it in place of `MoveToLayer(QWERTY)`, but no
    // longer do. We keep it so that if someone still has the old layout with
    // the macro in EEPROM, it will keep working after a firmware update.
    Layer.move(QWERTY);
    break;
  case MACRO_VERSION_INFO:
    if (keyToggledOn(event.state)) {
      Macros.type(PSTR("Keyboardio Atreus - Kaleidoscope "));
      Macros.type(PSTR(BUILD_INFORMATION));
    }
    break;
  default:
    break;
  }

  return MACRO_NONE;
}

void setup() {
  // Enable AutoShift for letter keys and number keys only:
  AutoShift.setEnabled(AutoShift.letterKeys() | AutoShift.numberKeys() | AutoShift.symbolKeys());
  // Set the AutoShift long-press time in milliseconds:
  AutoShift.setTimeout(185);
  AutoShift.enable();

  Kaleidoscope.setup();
  SpaceCadet.disable();
  EEPROMKeymap.setup(10);
}

void loop() {
  Kaleidoscope.loop();
}

```

</details>

**Expected behavior**
I expected this would enable AutoShift on my current layout. Additionally, since my keymap is stored in the EEPROM I expected my keymap to stay the same.

**Environment (please complete the following information):**
 - OS: macOS
 - Version: 11.2.3
 - Device: Keyboardio Atreus

**Additional context**
I noticed there was an additional plugin initialized in the example sketch. `AutoShiftConfig`. When I add that plugin in my sketch it causes my keymap to type the wrong characters and still doesn't seem to work with AutoShift.

I am happy to try anything else you might think of and would be delighted to beta test support for AutoShift and official support of AutoShift in Chrysalis. Thanks for your time and efforts!

gedankenexperimenter, 2021-07-03:
Regarding the stuck key, I have an idea of why that might happen (an error in handling a full event queue), but I can't test it at the moment. I'll get to it when I can.

obra, 2021-07-03:
M - do you think this is an autoshift issue or a core event handling issue?

gedankenexperimenter, 2021-07-04:
My suspicion is that the AutoShift queue is filling up, and it's failing to handle that properly.  I doubt it's a core event handling error.

gedankenexperimenter, 2021-07-04:
I've looked at the code, and there's definitely an error in how AutoShift handles the queue filling up.  I believe I have ~90% understanding of the problem, with high confidence that I will be able to put together a fix in the next day or so.  I still need to write the failing testcases that will prove I haven't misunderstood what's going on.

My intention is to change the behaviour slightly so that any new key toggling on will stop a long-press from becoming auto-shifted.  This should make it so that there's no need to increase the queue size.  This behaviour change is independent of the fix for the queue overflowing, but it's closely enough related that it probably doesn't make sense to use two separate PRs.  I will try to keep the commits separate, at least.

gedankenexperimenter, 2021-07-04:
@joe-sharp — I noticed that you have both AutoShift and Qukeys in your sketch. If you're using both plugins, I strongly recommend registering Qukeys first in `KALEIDOSCOPE_INIT_PLUGINS()`, because Qukeys will discard the changes that AutoShift makes to the event `Key` values if a qukey is being held.  Using both of these plugins on the same key(s) is dicey at best, regardless.  There is a similar issue with SpaceCadet.  If you're actually using all three of these plugins, the recommended order is Qukeys, then SpaceCadet, then AutoShift.  This should be noted in the docs for AutoShift.

joe-sharp, 2021-07-04:
@gedankenexperimenter - Thanks for the suggestion. I don't think I **need** Qukeys necessarily, I think it was just already in the Sketch I was modifying. I switched them and immediately AutoShift is working! I'll definitely want your extra fix for the ghosting issue, but this is totally usable now, thanks for the tip!

PR #1075 (2021-07-04): Fix AutoShift queue overflow problem
This PR corrects an error that would allow the AutoShift event queue to overflow, causing "stuck" keys.  There is now more robust protection against event queue overflow, and several documentation changes.

- Fixed a typo in the `enable()`/`disable()` docs
- Added some inline comments to clarify things
- Better explained the presence of AutoShiftConfig in the example sketch
- Add docs about proper ordering of AutoShift relative to Qukeys & SpaceCadet

In addition, I changed AutoShift behaviour to abort a pending AutoShift key if any other key is pressed whi [... cut]
closed 2021-07-04 by obra