keyboardio/Kaleidoscope #1092: Macros: A conditional step

algernon, 2021-11-25:
While thinking about keyboardio/Chrysalis#653, I had an idea.

One thing macros are very good at is assigning a key sequence to a single key. One case where that's incredibly useful is to input Unicode, or language-specific symbols on an otherwise US layout. However, for those to work properly, one needs to handle the case where shift is pressed, along with the default case where it isn't. As such, you can't - currently - encode that into a macro without having custom code.

To make that possible, what if we had a macro step that is itself just an `if ($condition) $stepA else $stepB` kind of thing? Where `$condition` would be one of the possible conditions provided by Kaleidoscope.

Encoded, this would look: `MACRO_ACTION_STEP_CONDITIONAL`, followed by a single byte, from a `MacroConditional` enum (see later), and then a macro step (ending with a `MACRO_ACTION_END` sentinel), and another macro step ending with the sentinel. We do not need an extra sentinel to end the conditional step itself. Any of the then/else steps can be empty (a single sentinel) too.

For conditionals, I'd initially go with something simple:

```
enum class MacroConditional {
  Custom,
  IsShiftActive
};
```

If the conditional is set to `Custom`, then we'll call an user-defined function, `bool macroConditional(uint8_t macro_id, KeyEvent &event)`, which can then decide what the custom condition is. We _might_ want to encode custom conditionals differently, however, to allow easier re-use: we could add an id byte if the conditional is custom, before the two branches, and then use `bool macroConditional(uint8_t macro_id, uint8_t conditional_id, KeyEvent &event)`.

This feels like a reasonable system, one that's extensible, and one that allows us to build shift-aware macros easier, among other things.

On top of this, we might want to have a `MACRO_ACTION_EXECUTE` step too, which just runs another macro, it's single byte argument. So we can save a few bytes for reusable stuff like doing the unicode input prefix and the like.

(We'll need the same stuff implemented for DynamicMacros too, of course.)

algernon, 2021-11-26:
Thinking further, there are two different approaches I can see: In one case, we encode two single steps for the conditional. In this case, we don't need the sentinel, because the step itself will either be fixed size, or be sentinel-terminated anyway. So we save two bytes there. In the other case, we encode multiple steps for each branch, in which case we do need a sentinel for both.

I'm leaning towards the latter, because that makes it easier to implement shift-supporting unicode input macros. And to make that even easier, we could add an unicode input step: `MACRO_ACTION_UNICODE_TYPE`, which takes a 32-bit argument, and simply `Unicode.type()`s it. But that adds a dependency on Unicode, which isn't something we want all the time. We'd also need a couple of other unicode-related step types, but lets not get there just yet.

However, there's a way to make it easy to input unicode via macros, that does not add a required dependency on Unicode! But it requires a redesign of the Macros plugin, which is larger in scope than this issue, so I'll describe it separately. The serialized format can remain the same, however, so we can implement the conditional step, and then work on the rest in another step.

jeancroy, 2022-04-30:
If this is for Unicode character input I'd like to see conditional on input method (or OS).

On linux there's either compose-key or ctrl-shift-u ,
On macos there's the special keyboard "Unicode Hex Input",
On windows there's both alt0-xxx and hex numpad.

   alt-0xxx is basically limited to latin-1 codepage, 
   while hex numpad support everything but works on at-most half the applications.
   
Maybe the gui is the correct place to choose an input method, but I really wish there was something more portable.