keyboardio/Kaleidoscope #1095: Macro does not work now

Myrddinlefou, 2021-12-12:
In previous version of Kaleidoscope I created macros and their work fine, but I try the newest version and now I cannot use my macro.
When I use Chrysalis and the initial firmware, macro 1 send the good result (Keyboardio Atreus - Kaleidoscope), but when I use IDE Arduino and the latest exemple of Atreus macro code, macro 1 send nothing. The compilation and the update of my Atreus is OK.

I do not understand the bug, and I have no information to correct it.

I see on keyboardio discord I am not the only one, and a solution is to remove Chrysalis, but I like the function of Chrysalis and I am not happy to remove it.
Did kaleidoscope's macro are remove by Chrysalis ?

I use IDE arduino 1.8.16, windows 10 and an Atreus keyboard.

gedankenexperimenter, 2021-12-14:
I can't reproduce this problem, using the latest Kaleidoscope & KeyboardioHID, and the example Atreus sketch, using Chrysalis to define a key as `Macro #1`: it produces the expected version info when pressed.

Myrddinlefou, 2021-12-21:
I jump to the latest version of kaleidoscope and now my macro work again.
Thank you for the test.

closed 2021-12-21 by Myrddinlefou