keyboardio/Kaleidoscope #1098: Build error for default Atreus firmware in Arduino IDE on Debian

FjolleJagt, 2022-01-15:
Hi,

I'm trying to get Kaleidoscope working on Debian Bookworm (Testing), but compiling the default firmware for the Atreus fails in the Arduino IDE (version 1.8.19). I followed [the instructions](https://kaleidoscope.readthedocs.io/en/latest/build_default_firmware.html) until the "Build the firmware" stage, then get the following compile errors:

```
Warning: platform.txt from core 'Kaleidoscope Keyboards' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory
Error compiling for board Keyboardio Atreus.
```

I attach the verbose compiler output: [arduino_errors.txt](https://github.com/keyboardio/Kaleidoscope/files/7875761/arduino_errors.txt)

obra, 2022-01-15:
Hi!

Did you install the Arduino IDE as a deb or as a download from Arduino? If
it's from a deb, the issue may be that Debian has unbundled standard,
required components of the Arduino platform.

FjolleJagt, 2022-01-16:
Thanks for the quick response! I installed it as a deb using apt, but I do have all the arduino-* packages installed, including arduino-ctags 5.8-arduino11-1, except for the package arduino-core which conflicts with arduino.

I forgot to mention yesterday that compiling the empty 
```
void setup() {}
void loop() {}
```
sketch works when I select the Arduino Uno board but not when I select the Atreus (with the same ctags error as before).

Happy to try installing the bundled version of arduino from their website if you still think that's the problem!

obra, 2022-01-16:
I really do suspect that that's the problem.
https://wiki.debian.org/Arduino tells me that they're playing all sorts of
games, attempting to build a DFSG-free version of Arduino, up to and
including not packaging Arduino's canonical toolchain for avr.

If things still fail with the actual Arduino package of the Arduino IDE,
I'd be delighted to help debug this further, but right now, this feels like
Debian making a principled stand for free software that turns out to be
somewhat user hostile.

PR #1099 (2022-01-16): Document that Debian's arduino package doesn't work
I was unable to compile the default firmware with the version of arduino that was packaged with Debian. On issue #1098, @obra explained that Debian's version of arduino has been heavily modified, and installing the version of arduino that's available on the arduino website fixed my compilation problems. I have therefore added a note to the docs suggesting not to use the Debian version of arduino.

Fixes #1098
closed 2022-01-16 by obra