keyboardio/Kaleidoscope #1100: A macro that prints the keyboard layout

Rom1deTroyes, 2022-02-01:
**Is your feature request related to a problem? Please describe.**

It would be handy to have a macro that prints the current layout of the keyboard and so helpfull when you are out with just a notepad to find an rarely user symbol !

**Describe the solution you'd like**

Once assigned to a key, press it, and the keyboard write all the keys one after an other, not like if you press all the switches from upper left to botton right, but as the laminated sheet with all the symbols (you don't want your key board to write <kbd>Ctrl</kbd>+<kbd>z</kbd> or <kbd>Alt</kbd>+<kbd>F4</kbd>, mmh ?).

**Describe alternatives you've considered**

The format could be in various format :
- `qwertyuiopasdfghjkl`
- On multiple lines :
  - <kbd>q</kbd> <kbd>w</kbd> <kbd>e</kbd> <kbd>r</kbd> <kbd>t</kbd> <kbd>y</kbd> <kbd>u</kbd> <kbd>i</kbd> <kbd>o</kbd> <kbd>p</kbd>
  -  <kbd>a</kbd> <kbd>s</kbd> <kbd>d</kbd> <kbd>f</kbd> <kbd>g</kbd> <kbd>h</kbd> <kbd>j</kbd> <kbd>k</kbd> <kbd>l</kbd>
- as a Kaleidoscope keymap so you can use Chrysalis to tweak your keyboard, and use you keyboard to tweak your ino sketch

**Additional context**
I don't know how much the firmware knows about his own geometry, neither how much of memory will be needed to acheive this feature, but it may be something outstanding !

It will also related to https://github.com/keyboardio/Chrysalis/issues/621 as a counterpart ?

algernon, 2022-02-01:
At the moment, the firmware knows the rows and cols of the keyboard, but not much else about its geometry. We'd need to encode a _lot_ of stuff into the firmware to make this possible, including knowledge of all the host-side layouts. For example, if your host is set to Dvorak, and the firmware keymap is set up accordingly, it can't print the layout assuming the host is QWERTY, because that'd come out wrong. All that knowledge in-firmware would be way too much. Assuming everything is QWERTY would drastically reduce the usefullness of it.

I think this is something where using a combination of firmware and host-side tools would be the best. That would require having additional software on the host, not just a notepad, but that software could help discovering hidden gems in the layout much better than anything we can do on the firmware side, in my opinion.

Rom1deTroyes, 2022-02-04:
> At the moment, the firmware knows the rows and cols of the keyboard, but not much else about its geometry.

That is what I suspect from the way the `KEYMAPS` are indented in the code (just a nice presentation :ok_hand:).

> We'd need to encode a _lot_ of stuff into the firmware to make this possible, [...]

Yep, and as an embedded system, place is not extensible :-(

Maybe an approach from the category of key is simpler :
- If it's a printable char, do as if the switch is pressed (your host will know how to handle it : Q or D)
- else, if it's a modifier or a macro, well, we think about what we can do (use a symbol ?)

The result will be a single line per layer, but it can be a good help when searching a symbol !

> I think this is something where using a combination of firmware and host-side tools would be the best. That would require having additional software on the host, not just a notepad, but that [...]

Yep, someone has developed a 💎 called Chrysalis that does this job perfectly well 🚀 

The main idea here should be having a simple macro/plugin not included in stock firmware, and usable everywhere, or for debugging.