keyboardio/Kaleidoscope #1113: MouseKeys not working with Macros

schmendrik, 2022-02-19:
**Describe the bug**
MouseKeys (mouseScrollUp, mouseScrollDn) do not work with Macros. While the sketch compiles just fine, nothing happens when a key triggering a macro using mouse keys is pressed. 

**To Reproduce**
Example implementation of macroAction:

``` c++
const macro_t* macroAction(uint8_t macroIndex, KeyEvent &event) {
  if (keyToggledOn(event.state)) {
    switch (macroIndex) {
      case MACRO_ZOOM_OUT:
        return MACRO(D(LeftControl), T(mouseScrollUp), U(LeftControl));
    }
  }
  return MACRO_NONE;
}
```
**Expected behavior**
The macro `MACRO_ZOOM_OUT` in the example code should send CTRL+MouseScrollUp to the OS, just as if I had pressed CTRL on my keyboard and scrolled up using my actual mouse wheel at the same time.

**Environment (please complete the following information):**
 - OS: Ubuntu Linux and Windows 10
 - Kaleidoscope Version 2022.2.18175535
 - Device: Keyboardio Model 01

gedankenexperimenter, 2022-02-19:
I've looked into this, and I understand the problem.  I've got a few ideas about how to solve it, starting with some reorganization of MouseKeys code, which I think is a no-brainer.  The bigger question is what to do about Macros.

MouseKeys works by searching the `live_keys[]` array every cycle for held mouse keys, and sending reports based on that.  The fundamental problem is that it doesn't know about the Macros virtual key array, so any mouse keys in a `MACRO()` (or `Macros.press()`) won't get noticed by MouseKeys.  We could change MouseKeys such that `Macros.tap()` (and, by extension, the above `T(mouseScrollUp)`) would work, but Macros would still not be able to emulate a _held_ mouse key.  We could accept this as a limitation (I think it's of dubious value, at best), and document it in MouseKeys, Macros, or both.

If we don't want to accept the limitation (after all, MouseKeys isn't the only plugin that might be affected), I think the best way to address the issue would be to add a new event handler or two (I haven't finished working this out yet), so that a plugin (e.g. MouseKeys) can ask "Do any other plugins have any virtual held keys? If so, please tell me about them."  Then a plugin that implements that handler (e.g. Macros) can call a handler implemented by the first plugin (i.e. MouseKeys), saying "I'm holding key `Key`?" (once per key value of interest).  So far, this is the only general solution I've thought of, and it has the benefit of not introducing any explicit dependencies on Macros.

gedankenexperimenter, 2022-02-21:
@schmendrik — I have submitted a solution as PR #1114, and tested it out.  With that change to MouseKeys, your Macros should work.  If you're able to check out that PR and test it that would be a big help.

schmendrik, 2022-02-24:
I've checked out the pull request, and did a quick test. It's working great so far. No problems or interference with my many other macros. If anything comes up though, I'll be sure to report it.

Thanks!

PR #1114 (2022-03-09): Restructure MouseKeys to be aware of Macros virtual keys
This change is designed to fix #1113 by restructuring how MouseKeys processes key events for mouse cursor and wheel movement.  Instead of only sending those reports in `afterEachCycle()` directly, I've shifted that code out to two new helper functions: `sendMouseMoveReport()` & `sendMouseWheelReport()`.  This allows us to both send periodic reports when a key is held, but also in response to key events directly, even when a mouse movement (or wheel) key toggles on and off within a single cycle.

However, because MouseKeys used to poll th [... cut]