keyboardio/Kaleidoscope #1163: Key_RightAlt returns event.key Alt, should be AltGraph

eriols, 2022-04-14:
**Describe the bug**
There seems to be some flakiness involved with using Key_RightAlt. 
In the majority of the cases it returns event.key Alt (event.which 18) whereas it should return event.key AltGraph (event.which 225).
Sometimes I get into a state where it returns 225, but I have not understood what causes this.

It's a rather annoying issue as I use AltGraph as the compose key on Ubuntu for all sorts of umlauts and no other key options are very appealing in being easy to access on both laptop and Atreus. Today I often find myself having to write "ä" using ctrl + u + e4, which is tedious.

**To Reproduce**
1 Map a key to Key_RightAlt and flash the keyboard, 
2 then use eg the site keycode.info to see which event it emits. 

**Expected behavior**
I expect Key_RightAlt to always return event.key AltGraph, not Alt. 

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04
 - Device Atreus Keyboardio

**Additional context**
Love the keyboard, I also have the old wooden version with more clicky keys but this one I use in the office. Thanks!

obra, 2022-04-14:
So there’s nothing in the firmware that should mix the two.  Is this reproducible on more than one computer? I recall there being something …special about ubuntu’s modifier handling in that era. 

Additionally, browser based event reporting can be a little bit dicey. Does xev report the same thing?

gedankenexperimenter, 2022-04-14:
It certainly sounds more likely to be software on the host causing this than something on the keyboard.  The only thing I can think of that seems even vaguely plausible if it is the keyboard is the key in question being mapped to `Key_RightAlt` on one layer and `Key_LeftAlt` on another, which is accidentally getting triggered sometimes.

eriols, 2022-04-14:
> So there’s nothing in the firmware that should mix the two. Is this reproducible on more than one computer? [snip] Does xev report the same thing?
Thanks a lot for the suggestions and hints. I regret to say I can't try it out more for a few days since I'm out of town for a few days. Will report back early next week.