keyboardio/Kaleidoscope #1168: Non-functioning under coreboot + seabios, model01

ian-kelling, 2022-05-13:
**Describe the bug**
Keyboardio Model01 using kaleidoscope under coreboot + seabios plugged in to a kgpe-d16 motherboard does not work.

**To Reproduce**

Reboot computer. Coreboot loads seabios. Pressing keys does nothing. Also, LEDs do not light up, or only do briefly. Typing on a cheap mass market keyboard will sends input, eg: pressing escape boot menu selection, ctrl-c to cancel PXE dhcp wait periods.

**Environment (please complete the following information):**
 - OS: coreboot git version 7bdedcdc338e5043f9670790, which is in the 4.10 range, with seabios payload, using default options related to keyboard
 - Device: Keyboardio Model 01
- Kaleidoscope:  git version 7d08de5d239cf2, which is may 8th. The model01 sketch has mostly some basic customization.

**Additional context**

Any keyboard which gets plugged in after coreboot has started up won't work, it has to be plugged in from the beginning.

I tried running the USBQuirks fix from a keybind, and from the start of the sketch, and and removing NKRO support entirely as described https://community.keyboard.io/t/no-input-in-coreboot-bios/5299/4, none of that made any help. 

I also noticed that LED functionality mostly does not work as expected under coreboot. For example, with kaleidoscope::plugin::LEDSolidColor I got the led to flash briefly sometimes but never stay on. I can work on testing that more.

Rom1deTroyes, 2022-05-13:
The leds part make me think of a power consumption problem : my Model01 led patterns work well even when plugged on a powerbank (with no computer)...
... On the other hand, there were reports of keyboard problem with some BIOS configurations...
... And of course problem with over current or voltage drop from the hardware and OS limitation, or suspending the USB ports when sleep mode wakes up.

A test can be to plug the 2 keyboards on a powered USB hub, and see if they works ?
Or having an USB Current / Voltage Detector to know if there is a power limitation on the USB ports of the motherboard ?

obra, 2022-05-13:
The "all keyboards need to be plugged in before you boot" thing suggests to
me that there's something weird with the implementation.

ian-kelling, 2022-05-14:
Your comments make me think it would be good to be able to see what data and power the computer is sending. I think those require some special equipment to be between the computer and keyboard. I will look into options and any suggestions are welcome.

gedankenexperimenter, 2022-05-14:
Possibly relevant: keyboardio/KeyboardioHID#50

That PR is very out of date now, and was only ever intended as a means to get a better understanding of how things worked with boot protocol _vs_ NKRO protocol.  Unfortunately, I don't think I ever succeeded in getting anyone to look at it.  Maybe it will lead to a solution to this problem, though.  I still don't understand how what I did in that PR worked, but it did.

algernon, 2022-05-17:
I think that PR will be our solution here. I _think_ I know why it works, but I'll test it on a bunch of hardware to make sure.

ian-kelling, 2022-05-17:
I am happy to test it, would someone please port it forward to the latest git version? It doesn't build against master and the setup building has also changed, so it also isn't just a matter of checking out an older version. algernon may already be porting it forward, so I don't want to duplicate his work.

algernon, 2022-05-17:
I haven't started porting yet, and likely will not be able to in the next few days, so if you have some time, and are willing to, please go ahead.

ian-kelling, 2022-05-18:
I ported the patch, except the serial output. It didn't help get bios input. I also tried setting the default to boot protocol since the keyboard cant be replugged in, I'm skeptical the usb quirks method would work. That didn't help get bios input either. https://github.com/ian-kelling/KeyboardioHID/tree/iank-boot-protocol

obra, 2022-05-18:
Are you set up to debug from the seabios side? it looks like they have some
debug print functionality:
https://github.com/coreboot/seabios/blob/master/src/hw/usb-hid.c

It also sounds like their stuff is moderately fragile?
***@***.***/thread/J6FHNS6H76FUR7EF5HTLMWQSXW5A7OIO/

What's the easiest path for me to get a working seabios environment for
testing? I've got the requisite USB protocol debugging gear.

ian-kelling, 2022-05-18:
Yes, I can build and debug seabios. For you to work on it more directly: it depends on if you need a machine physically there with the USB protocol debugging gear. 

Remotely: I could setup a machine for you to test with ssh + serial connections. There is serial output from kgpe-d16 + seabios and the model01 has serial output. So, you'd ssh to the machine under test, flash new seaboot/kaleidescope, reboot, ssh to 2nd machine which has serial usb adapters, interact with seabios + keyboard over serial, repeat. If you flash a non-booting image, it only takes a few seconds for me to fix manually, so I wouldn't bother trying to automate it: The bios chip is in a socket which you can pull out by with your fingers, so I just swap in a chip that has a good coreboot image, boot, once linux takes over, swap back to the nonbooting chip, then you can flash a new coreboot+seabios that you hope will boot and flash it with `flashrom -p internal -w ROM_FILE`.

If you need the hardware in front of you, an x200 laptop with seabios is probably the easiest. I could send you one, I have spares (I assume this isn't specific to d16 motherboards, I would test before you go with this option).

gedankenexperimenter, 2022-05-21:
> I am happy to test it, would someone please port it forward to the latest git version? It doesn't build against master and the setup building has also changed, so it also isn't just a matter of [...]

I haven't tested this yet, but I went ahead and rebased my original boot protocol switching PR onto KeyboardioHID master here: https://github.com/gedankenexperimenter/KeyboardioHID/tree/fix-protocol-switching

@algernon — If you'd like, I can force-push those changes onto keyboardio/KeyboardioHID#50.  I hesitate to do so because I don't want to complicate anything you might have in progress.

algernon, 2022-05-21:
You can go ahead and force push, it won't complicate anything for me. Thanks!

obra, 2022-10-11:
To really debug this, I'd need to have the machine on my desk so I could
stick my Beagle480 USB analyzer in between the keyboard and the computer.

It absolutely makes sense to try a keyboard with *just* the BootKeyboard
device enabled before going that route.

obra, 2022-10-11:
> Kaleidoscope: git version 7d08de5, which is may 8th. The model01 sketch has mostly some basic customization.

Can you point us to docs about coreboot’s USB keyboard support? Absent a test device, it may be tricky to figure out.

ian-kelling, 2022-10-12:
Sounds like I should just mail you a machine. I have ones that have defects like broken ethernet but area great for keyboard testing. Email me an address to send to, ian@iankelling.org. I live in the Boston area.