keyboardio/Kaleidoscope #1169: DynamicMacros revamp

algernon, 2022-05-21:
While [working on Macro support for Chrysalis](https://github.com/keyboardio/Chrysalis/pull/703#issuecomment-1133628536), I ran into a few shortcomings of the way DynamicMacros currently work. The major problem is that it uses different keycodes than normal macros, which makes it awkward in Chrysalis to assign them, and adds complexity too, because one has to keep in mind which macros are built-in, and which are dynamic.

Instead, we should do what DynamicTapDance does, and sit on top of the existing implementation, to the extent possible.

This would give us a number of advantages:

- Less confusion, because Macros are Macros, regardless of where they are stored.
- More dynamic macros! At the moment, we can have at most 32 of them. If we reused the Macros keycodes, we could have a total of 256 macros, potentially all of them editable.

The downside is that the DynamicTapDance setup needs some assistance, and the `macroAction` callback would need to call into `DynamicMacros` when appropriate. I don't think that's a big deal, especially since we can ship default firmware in Chrysalis (and on new boards) that handle that case for us. Users with custom macros can add a few lines, too. Basically, replacing the final `return MACRO_NONE;` with `return DynamicMacros.play(macro_id);` would do the trick pretty much in every case.

algernon, 2022-05-22:
There's one thing we need to note here: we have a range reserved for dynamic macros at the moment, which will become obsolete with the proposed redesign. While we usually keep backwards compatibility, I feel that in this case, we don't need to. We'll still keep the range reserved, because we do not want to move the stuff that comes after. But it will no longer be used, and we may reuse it later on.

gedankenexperimenter, 2022-05-22:
On a related note, the existing Macros range should really be moved so that it's a subset of the plugin range, rather than a part of the built-in range.  Maybe we should transition with a few versions of Kaleidoscope and Chrysalis recognizing both, but only writing the new range?

algernon, 2022-05-22:
> On a related note, the existing Macros range should really be moved so that it's a subset of the plugin range, rather than a part of the built-in range. Maybe we should transition with a few versions of Kaleidoscope and Chrysalis recognizing both, but only writing the new range?

Hm, that's a sensible idea, recognising both, but writing only from the new range. I'd tackle that in a separate issue, though, even if we do both at roughly the same time. Mostly so I don't get lost. Can you open an issue, please? Thanks!

algernon, 2022-05-22:
Looking over DynamicTapDance, I'm not happy with that, either.

The problem is how we handle the mixing of built-in and dynamic macros (or dances; we'll focus on macros here, but this all applies to both). With DynamicMacros, we have two entirely separate sets, which makes the Chrysalis parts a little awkward, and it feels silly, too. Whether built-in or dynamic, a Macro is a Macro.

However, moving over to use the Macros keycodes is not without downsides. It's best explained by a scenario: suppose we have two built-in macros, and four dynamic ones. We have `macroAction()` set up to call `DynamicMacros.play()` for the 4 dynamic macros.

The problem is from an UI point of view: to save space, we want to start saving dynamic macros at id 0, even if we have built-in ones, yet, `macroAction` will play our first dynamic macro with macro id 2 (the two builtins are 0 and 1). When we edit `Macro #1` in Chrysalis, we're editing the 2nd dynamic macro, rather than the 2nd macro total. To counter that, we can tell Chrysalis how many builtins we have, so when trying to edit `Macro #1`, Chrysalis would simply say "Sorry, it's a built-in macro". But then we need an extra byte to hold this offset, and an extra focus command.

Hmm. Maybe that's not so bad. It's better than the alternatives I considered, like encouraging users to place their macros at higher IDs, because in the firmware, that comes at no cost, and then we could edit dynamic macros in Chrysalis freely. We'd still have a situation where we reach the built-in area, but! Because we keep an in-memory macro id -> offset map, DynamicMacros is currently limited to 32 macros. We could increase that, but then we'd increase RAM usage too. Not a problem on GD32 where I expect DynamicMacros will be much more useful, but a problem on AVR.

With this in mind, we could split the existing Macro range:  the first 32 would be DynamicMacros, and built-ins would start after. That's not much better than the first option outlined above (storing an offset), we just made the offset hard-coded, and in a different direction.

So, TLDR: the offset method is still the best. Chrysalis can present macros at their real id, show "This is built-in, no can do" for builtins, edit dynamic ones, and offset them at storage time.

obra, 2022-05-22:
Don’t forget that for Chrysalis, we want to hide the macro id for dynamic macros from the user. Just “there is a 
macro on this key”

algernon, 2022-05-22:
Well, we kinda need to distinguish macro keys somehow, even in Chrysalis. And if we re-use the built-in macro ids, then we need to know the ID. It won't mean much, but it's an indicator.

obra, 2022-05-22:
it makes sense to me to expose them for built in macros but i explicitly don’t want the indicator for dynamic macros.

algernon, 2022-05-22:
Can certainly give that a try and see how it feels.

algernon, 2022-05-22:
We discussed this further with @obra, and arrived to the conclusion that in the end, we do want to keep the DynamicMacros range separate from the built-in macros, because we can build a better, easier to use, more consistent UI on top of that.

As such, I'm closing this.

closed 2022-05-22 by algernon