keyboardio/Kaleidoscope #1170: Migrate the Macro keycode range out of the built-in range

algernon, 2022-05-22:
For historical reasons, Macro keys use keycodes within the built-in range, which is undesirable, because it makes it hard for us to distinguish core keycodes from plugin keycodes, and makes `kaleidoscope::ranges` a bit weirdly shaped, too.

As [suggested](https://github.com/keyboardio/Kaleidoscope/issues/1169#issuecomment-1133909840) by @gedankenexperimenter, we could do a gradual migration: recognise both ranges on the firmware side temporarily, but newly compiled firmware and Chrysalis would only assign from the new range.

Over a longer period of time, that'd eventually fade the old ranges out, while maintaining 100% compatibility. Chrysalis could even auto-rewrite eeprom keymaps, or offer to do so like we do for some legacy keycodes, to make the transition smoother.

gedankenexperimenter, 2022-05-23:
Perhaps the best thing would be for Chrysalis to have some way of querying Kaleidoscope to find out what version of the firmware is installed, so I knows which keycodes to use for Macros keys.

algernon, 2022-05-23:
> Perhaps the best thing would be for Chrysalis to have some way of querying Kaleidoscope to find out what version of the firmware is installed, so I knows which keycodes to use for Macros keys.

While that'd be great, it isn't trivial (or even possible) to implement it in a way that Chrysalis can 100% rely upon it. We could definitely do it for firmware shipped with Chrysalis, but for custom firmware, it's trickier.

I think it is less effort to show a prompt asking the user whether we should migrate, and the minor annoyance is perhaps acceptable.

gedankenexperimenter, 2022-05-23:
> I think it is less effort to show a prompt asking the user whether we should migrate, and the minor annoyance is perhaps acceptable.

Sound good to me.