keyboardio/Kaleidoscope #1177: DynamicMacros fails to properly parse empty macros

algernon, 2022-05-24:
While testing the macro support for Chrysalis, I ran into an issue: if I have empty Macros, Kaleidoscope fails to parse them properly, by the looks of it.

The scenario is this: I created 7 macros, with the last one being a "Hello World!". While all 7 had at least one step, the last macro worked fine. When *any* of the previous macros is an empty one (just a zero byte in storage), the last macro doesn't work. Adding a dummy "wait 0" to previous macros restores the functionality.

We do need to support empty macros, so indexes don't jump around when we empty out macros in Chrysalis.

algernon, 2022-05-24:
Same problem doesn't exist for Macros, because those are stored differently, and we do not need to split up continous storage space to find macro boundaries.

gedankenexperimenter, 2022-05-24:
I assume this is just macros that use `Macros.play()`?  Could it be that `MACRO_NONE` and `MACRO_ACTION_END` are indistinguishable (both `0`)?  I'm just speculating based on a very brief glance.

algernon, 2022-05-24:
It's probably that. On top of that, we don't handle `255` either, so `macros.map 255 255` will lock the keyboard up.

algernon, 2022-05-24:
The problem was that the following serialized sequence wasn't parsed properly: `0 0 8 4 0 255 255...`. The intention here is to have 3 macros: 2 empty ones ([0], [0]), and one with a single step ([8, 4, 0]). An empty macro is one that has no content, and is just a terminator.

The first `0` is parsed correctly, because `previous_macro_ended` starts off as false, so we don't trigger the return. We do set the same variable to true, though, and then we encounter the next `0`, and return, because `previous_macro_ended` *is* true at that point, because there were no other steps inbetween. Thus, we miss the `8 4 0` sequence, which would be out third macro: `TAPCODE A, MACRO_ACTION_END`.

gedankenexperimenter, 2022-05-24:
That was my understanding; thanks.  But with the existing code, if you had a working set of macros, then changed it, the higher-numbered ones might still do _something_, but not necessarily what they were intended to do.

closed 2022-05-24 by obra