keyboardio/Kaleidoscope #1190: Document how to use the Arduino IDE with git master

algernon, 2022-05-30:
Way back when, it was possible to clone the [Kaleidoscope-Bundle-Keyboardio](https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio) repository into `~/Arduino/hardware/keyboardio`, and work with that, as is, with Kaleidoscope being under `avr/libraries/Kaleidoscope`. I could symlink that to another place to make it easier to work with it, but it still worked as-is.

Nowadays, we need to move `plugins/` to a place where Arduino can find them. That's done by the `bin/build-arduino-package` script, but that does a whole lot of other things too.

It would be great to have some official documentation - and perhaps a shorter script than `build-arduino-package` - to help set things up.

obra, 2022-05-30:
It's not currently possible to git clone Kaleidoscope and have it usable in
the Arduino IDE. That's why we're publishing packages on every commit.
https://github.com/keyboardio/arduino-kaleidoscope-master

This is actually documented in our "how to set up the Arduino IDE" docs at
https://kaleidoscope.readthedocs.io/en/latest/setup_toolchain.html

algernon, 2022-05-30:
Yeah, I read those docs. They're great when someone wants to use the latest Kaleidoscope. It's not very helpful when someone wants to *develop* Kaleidoscope, and still use the IDE, to make sure things work in that environment too.

From the commandline, this is no problem, because we can just pass `--libraries "${KALEIDOSCOPE_DIR}/plugins/` to `arduino-cli`, and all is well. That doesn't work for the IDE, we need to do a symlink farm, or copy things over. That's the part we should automate, in my opinion.

cxw42, 2023-07-02:
Any tips you can provide would be very helpful!!! I last updated my Model100 firmware last year and can't for the life of me remember how I did so :( .  I have looked at readthedocs, searched the forums, read READMEs and Makefiles, and am still not certain how to proceed after <https://kaleidoscope.readthedocs.io/en/latest/build_default_firmware.html>.  There seems to be a missing step:
[image]
At the arrow, I think it should say "how to set up your sketches/libraries/directories to use your own copy of Kaleidoscope."

(Related: for local development, do I clone Kaleidoscope, Kaleidoscope-Bundle-Keyboardio, or Chrysalis-Firmware-Bundle?  The docs suggest to me I should use Chrysalis-Firmware-Bundle, but it's not pinned on <https://github.com/Keyboardio>.)

cxw42, 2024-03-10:
I was able to update FocusSerial for my Model01 today (https://github.com/keyboardio/Kaleidoscope/issues/1235#issuecomment-1987325754) by:
1. Cloning Kaleidoscope (in my case, into ~/proj/Kaleidoscope --- use whatever dir you want)
1. Setting up the Arduino IDE to use the Model01 board
1. Moving the old libraries:
  ```
  cd ~/.arduino15/packages/keyboardio/hardware/avr/1.99.8/libraries
  mkdir ORIG
  mv * ORIG
  ```
4. Symlinking the cloned Kaleidoscope into its place:
  ```
  for f in ~/proj/Kaleidoscope/plugins/* ; do ln -s "$f" . ; done
  ln -s ~/proj/Kaleidoscope .
  ln -s "$(realpath ORIG/KeyboardioHID)" .
  ```
5. Opening the Model01-Firmware sketch in the Arduino IDE
6. Building and flashing as usual 

Please, please tell me there's a better way! :laughing: :sweat_smile: