keyboardio/Kaleidoscope #1195: Rainbow LED schemes on Model 100 disable pgup/pgdn

rhomboid-shovel, 2022-06-04:
**Describe the bug**
On a Model 100 with very minimal keymap changes (remap layer 2 for S, D, F, and E and remap layer 0 for num), two of the LED modes seem to disable pgup and pgdn. The two LED modes are the full-board rainbows - #1 is single color across entire keyboard, and that single color shifts through the rainbow; #2 is entire rainbow spectrum across entire keyboard and the rainbow shifts across the keyboard.

**To Reproduce**
Hit the "led" key until you are in one of the two full-board rainbow modes, and pgup/pgdn stop working. So do their layer2 functions, home and end. As far as I can tell, including using a keyboard tester program, pgup/pgdn are not doing anything in these two modes. 

If you hit the "led" key until you are in any other mode and pgup/pgdn begin to work again. 

**Expected behavior**
For pgup/pgdn to act as pgup/pgdn regardless of the LED mode.

**Environment (please complete the following information):**
Keyboardio Model 100 with whatever version of Kaleidoscope shipped on the boards.
Win10, for what that's worth

obra, 2022-06-04:
If you flash your firmware with the current build of  Chrysalis, this
*should* be resolved.

rhomboid-shovel, 2022-06-04:
That's all, and after I isolated the bug and wrote this gem of a bug report? Okay, once I can get Chrysalis working again, I'll try flashing the latest firmware.

rhomboid-shovel, 2022-06-09:
Fixed in 0.10.2. Thanks, friends!

closed 2022-06-09 by rhomboid-shovel