keyboardio/Kaleidoscope #1204: Weird "C" key behavior on Model 01

mentalisttraceur, 2022-06-13:
**Describe the bug**
Often, if I hold down the "C" key (the physical key on the Model 01) long enough, I get this weird behavior where

1. a bunch of keys to the right of it light up white (not always the same ones either - see attached pictures) and will stay lit up as long as I hold C, and then
2. in some subset of those times (I haven't figured out a pattern for when it will happen), **after I let go of the C key**, the key directly above it ("D") will act as if unbound (pressing it causes seemingly no input to the computer), until I do other stuff on the keyboad (I haven't figured out a pattern for what exactly causes it to fix itself).

I normally don't hit this problem when typing, but it can be seriously unusable in some edge cases (for example, a video game where you might hold down the C key to use an ability, and then you try to use another ability on the D key and nothing happens).

I kinda suspect that this could be a hardware problem rather than a Kaleidoscope problem, but the situation is extremely search-engine-unfriendly and I just haven't been able to find anything about it either way, so I'm posting here in the hopes that someone recognizes it and can point me in the right direction.

**To Reproduce**
Hold down the C key for at least a second. If it doesn't happen on the first try, do it a few times. Maybe try again later after doing stuff if it still doesn't work - I can't tell if it's entirely random or if there's some sort of rules to it.

**Expected behavior**
Holding down the C key just works like holding down any other key.

**Environment (please complete the following information):**
 - OS: Happens on both Linux (PureOS distro) and I think Windows 11 (I'll edit when I've had a chance to double-check on Windows).
 - Version: Happened with both the latest firmware automatically found by Chrysalis (I'll edit later with the exact version), and the original out-of-the-box firmware from years ago).
 - Device: Keyboard Model 01

**Additional context**
Here are pictures of separate instances of the bug (note that different amounts of keys light up white each time, with some observed bias towards the first version where all columns to the right do):

[image]
[image]
[image]
[image]

mentalisttraceur, 2022-06-13:
Also often when this happens it seems like the same wrong scancode sequence is sent to the computer, at least once, something like "3e.e" (I'll edit again later when I have a chance to double-check exactly what it is.)

gedankenexperimenter, 2022-06-13:
Is your keyboard's firmware up to date?  If not, I strongly recommend updating it (Chrysalis is easiest, if you don't build your own firmware), and seeing if that changes things.  You also might want to try the hardware test mode.

obra, 2022-06-13:
This sure sounds like some kind of hardware failure, though I can't imagine
what it might be.

Best,
jesse

mentalisttraceur, 2022-06-14:
I've updated my prior comments with the exact key sequence that it sends when this happens (which makes more sense now), and now I had time to get the exact firmware I had and also try with the latest.

And I have a few more datapoints!

1. Sometimes after it gets into this state, besides making some button not respond to presses, it will just keep acting as if some other button is stuck pressed down (usually the one directly above, the D key, but sometimes the one at the top of the column, the 3),
2. The "reset' near as I can tell after it gets into this state is to hit the D key and then the 3 key, but I'm still not sure if that's 100% reliable or if sometimes there's more to it than that, or if that's not necessarily necessary.
3. In the moments that it sends a combination of scancode, it's predictably the ones for the 3EDC column, in that order (so since I have Dvorak set up it typically emits "3.ej"). Almost as if the electronics for the keyboard get shorted somewhere (presumably physically near the C key) somehow and so the circuitry fires as if I'm hitting everything in that column within the same clock cycle or in a really quick cascade. (I'm assuming stuff like LED controls are wired somehow in series, so maybe the reason all or some of the keys to the right go white when this happens is because the signal further through them is somehow attenuated/choked/etc due to shorting or greater resistance or something?

@obra Jesse, you're one of the two people behind keyboard.io, right (if I am recognizing the name correctly)? Do all the updates that it seems to have most of its effects in that same key *column* tell us anything useful? Should I be opening it up and looking on the PCB for signs of something spilled/smeared/deposited around there?

mentalisttraceur, 2022-06-14:
Aww this note on the PCB is adorable ❤️:

[image]

Anyway, I gave it a careful look between the layers and from the top of the C key's switch and on the bottom of the PCB, but I guess I can't pull it apart any further without soldering?

@obra would you be interested in me sending it back to keyboard.io for you to investigate more hands-on? (I don't remember anything about the warranty and I'm not much of a warranty guy, but I'd be willing to send it even without warranty replacement just if you wanted to look at it.)

mentalisttraceur, 2022-06-14:
@gedankenexperimenter hardware test mode is huge!

I can't seem to get GitHub to cooperate with me attaching a video to this issue just yet, but I'm seeing SO MUCH useful information from matrix test mode now.

For example, when the bug triggers, all four keys in that column FLIP OUT (they get in a state where all of them keep flickering green and... yellow? then when I get a key press to "reset" it to silent, all four in that column are obviously red - that freakout they did presumably gets counted as chatter).

Also I don't remember if I said this already but it's actually not needed for it to be a long press for it to misbehave like this! It'll just do it from a normal tap sometimes too.

Taking a moment to look at the "procedure to remedy key chatter" stuff. I'm not sure if I've got an MP1 and MP2 keyboard, but maybe it'll help.

(I guess if there's a determinism to the force/depth/angle of pressing C and whether the problem triggers, I can also use hardware test mode to get better visual feedback to my brain to train it, so maybe I can still salvage this thing through sheer persistence that way even if all else is lost.)

tlyu, 2022-06-14:
I've seen similar LED glitches when the PCB has gotten wet (due to traces of water from the isopropanol I used to clean keyswitches). The LEDs "downstream" of a given LED start to flicker or stop receiving updates. I suppose it could also be an individual LED's data-out line becoming flaky, but if the symptom doesn't always start with the same LED position, I think that would tend to indicate a wiring or PCB problem.

The LEDs are chained in a serpentine fashion roughly in columns (see the comments in https://github.com/keyboardio/avr_keyscanner/blob/master/firmware/led-issi31fl3743b.c for some maps), and your photos do make it seem like it's all the "downstream" LEDs that are misbehaving.

If the key event problems and LED problems seem to be caused approximately simultaneously by the same stimuli, that makes me wonder if there are multiple bad solder joints (or broken traces, but those are harder to fix) in the vicinity of that key. I think if it happens more often when you hold the key down more forcefully, that would tend to reinforce this hypothesis.

mentalisttraceur, 2022-06-14:
@tlyu yes! exactly this:

> I think if it happens more often when you hold the key down more forcefully

(and seemingly more readily for some time _after_ I hold the key down more forcefully).

Part of why I initially reported that this happens on long-press is that I tend to naturally bottom out when I hold a key down, but much less so during normal typing if the key gives enough tactile feedback, and it seems like using more force to quickly bottom out is part of more reliably reproducing the problem. 

I'm really glad to hear someone else has seen behavior like this.

mentalisttraceur, 2022-06-14:
!!! FIXED IT (not really but close enough) (for now... but hopefully forever) !!!

I followed the [alcohol cleaning instructions for solving key chatter](https://community.keyboard.io/t/what-to-do-if-your-keyboard-seems-to-duplicate-keystrokes-or-a-key-stops-working/636) and it's **way better** now!

Well, I must confess I was super clumsy with the isopropyl alcohol, so I probably had some leak around the edges of the switch and down onto the PCB. So I also gave it over an hour to dry, maybe closer to two.

**Before**: I was getting to the point where I could reproduce the issue almost at-will, within a few bottoming-out long keypress tries at most, but sometimes ever with fairly fast presses.

**After**: I have to be deliberately needlessly forceful and almost *roll* the key during my keypress to cause the issue, and the three times I've managed to reproduce it, way less chained LEDs lit up two of the times during which the entire 3EDC column remained totally fine/clean/stable, and only one of the times (when the largest amount of LEDs went white as in the first picture) only then did the 3EDC column freak out with rapid persistent spontaneous chattering ghost keypresses.

Also I may have reproduced it by accident a fourth time after leaning on the keyboard with my palm or forearm during some other fiddling, because at one point after fiddling with stuff further on my table, I looked back and and the 3EDC column was lit up red (chatter detected - this was in hardware testing mode), but nothing was flickering and inputs behaved correctly after.

mentalisttraceur, 2022-06-14:
The most interesting part I've noticed so far is that reproducing the issue now seems to require a "roll" motion. Like I bottom out but pushing rather angled to the left, and then I roll the push towards me. Make of that what you will.

I'm starting to think that I could learn to feel the the tactile feedback of the Model 01 switches, and if I stay within the sweet spot of not bottoming out or rolling roughly on the C key, then I might never trigger this bug.

I'm hesitant to keep re-producing it -  I don't know how much of this is possibly affected by messing with it, so I don't want to worsen it again by deliberately pushing it to the point or reoccuring.

mentalisttraceur, 2022-06-14:
Thank you each: @obra - you helped me more confidently collapse onto thinking about the problem as if it was probably a hardware issue; @gedankenexperimenter - you put me onto hardware testing mode and that turned out to be profoundly helpful, as well as linked to the chatter fixes; @tlyu - you helped me confirm some of the more detailed stuff and gave me more confidence that the more-forceful stuff was actually a factor rather than just something I was reading into it.

mentalisttraceur, 2022-06-14:
I still would love to hear any thoughts on the issue from anyone else.

I'll leave this open for another day or two and give this a lot of fairly extensive testing to see how it works out - if it's still holding up well after that, maybe it's worth pinning this issue in case some other person has the misfortune of developing an issue like this.

tlyu, 2022-06-14:
@mentalisttraceur now that I think about it, two of your symptoms are more likely the result of short circuits rather than open circuits. LEDs turning white suggests a LED data line short to VLED (white = all ones). All keys in a column acting like they're pressed suggests a column line (or R3C3 keyswitch pin?) shorted to ground, if I'm reading the keyscanner code correctly. There are probably ways to confirm (or distinguish) these relatively noninvasively (I _think_ all the relevant test points are exposed on the back of the PCB?), if you have a multimeter you trust to not use too much voltage or current during resistance tests. You might need a helper or to fabricate a test jig (or solder wires to test points) to hold the probes to the test points while you apply force, though.

@obra this is really sounding like a hardware problem to me. Do you prefer that we move to the forum? I guess I could also speculate about ways to make the firmware more robust or enhance its diagnostic capabilities (flipping the drive direction of the keyscanner, for example).

obra, 2022-06-16:
It does sound like a physical short circuit probably related to physical
stress on the board. Since it sounds like that one switch may be at or near
the culprit, the simplest thing might be for someone handy with a soldering
iron to desolder the switch, let you test the board and then either solder
it back in place or replace it.

@tylu - Better diagnostic capabilities in firmware would certainly be
interesting. Happy to have that chat either in GH issues or over on
discourse. :)