keyboardio/Kaleidoscope #1211: Kaleidoscope should not permanently lock up on incomplete Focus traffic

algernon, 2022-07-04:
Many of our Focus commands have a chance to lock up the keyboard permanently (until a reboot) when they're sent invalid or incomplete data. We shouldn't do that. We should have a timeout, after which the focus command is stopped being processed, and we pretend nothing happened.

tlyu, 2022-08-27:
Further investigation with a Model 01 (though admittedly with newer firmware that might mitigate the worst of this behavior) shows that the lockup isn't actually permanent. A timeout _does_ eventually occur if a different keyboard is used to press Enter to send the command (one second per call into a timed `Stream` parsing method, which is the default timeout).

However, if autorepeat is set fast enough, and the keyboard being commanded is itself used to press Enter to send the command, the keyup event on the release of the Enter key will be delayed until the Focus plugin returns, which won't happen until CR or LF characters stop being sent by the effectively held-down Enter key. Switching windows will send the held-down Enter keypress elsewhere, allowing the plugin to time out.

Tested with both `screen` and the Arduino IDE serial console on macOS.

tlyu, 2022-08-28:
I realized another barrier to accomplishing this without major redesign of Focus: once Focus dispatches the command, which can happen on the first Space character, the keymatrix isn't scanned until the plugins complete execution of the command. This means that it won't be possible to interactively enter a command with arguments, at least not on the keyboard that's being commanded.

Reading an entire line before dispatching will consume lots of RAM for commands that take a lot of input. I guess it's possible for Focus to do some or all of the argument parsing before dispatching, but I guess that requires a new API that allows a plugin to declare what formats it uses for its arguments. Alternatively, allow for some kind of way for a plugin to asynchronously accumulate arguments, but that raises the barrier for third-party plugin authors who are unaccustomed to an asynchronous programming style.