keyboardio/Kaleidoscope #1212: New hook proposal: onStorageSizeQuery

algernon, 2022-07-04:
It comes up over and over again that having a way to query the storage layout, or at least slice sizes of select plugins would be great. 

It all started in #344, which was quite ambitious, and wanted to implement a framework that'd let us query the Storage layout with little work from the participating plugins. At the time, the idea was that knowing the layout would allow Chrysalis to rearrange the layout if need be, when going from one firmware version to another. Since then, Chrysalis does structured backup & restore, which handles essentially the same thing in a simpler way. Thus, we no longer need to know the exact layout of the storage. It could be useful in some cases still, but much less important.

Then in #1206, it came up again, because we  have a plugin there that has an output format that doesn't make it clear how big the storage size is, unless we either bolt the size on at the end, or we provide a second Focus command to query the size alone. Neither of those feel appropriate.

So here's another proposal: a new hook, `onStorageSizeQuery`, which would be opt-in, for plugins that do not otherwise expose the storage size they use, and where knowing that size is important. At the time of this writing, that'd only be the `LayerNames` plugin proposed in #1206, but more may follow.

There are a number of other plugins where we want to know the storage size (Colormap, EEPROMKeymap, DynamicMacros, DynamicTapDance), but they expose the size in different ways: the data they return is a dump of the full storage, so if we count the bytes, we know the storage size, and them opting in to `onStorageSizeQuery` is unnecessary. They still could, but it'd be a waste of space. I'd rather have the host do the heavy lifting, if at all possible.

The hook itself would provide a very simple output: `pluginName storageSize`, one plugin per line. I'd call the hook from `FocusSettingsCommand`, when handling the `storage.size` focus command.

algernon, 2022-07-04:
> Do these plugins not need to register slices with the eeprom core?

They do need to, and they register slices with the core, yes. But we have other means to obtain the slice size in those cases (because they always dump their full storage on query), so an extra size query doesn't buy us anything, but it costs progmem. The LayerNames case is different, because it does not dump the full slice (hence the `0 <size>` entry at the end, which is what this plugin would replace).

obra, 2022-07-08:
What I was trying to get at was that, at least in theory, this and 'requestSlice' could replace each other. If we implement this hook, could we replace all requestSlice calls with an onStorageSizeQuery loop?

algernon, 2022-07-08:
I'm not seeing how they could. `EEPROMSettings` keeps track of how much space has been allocated, and `requestSlice` simply returns the current storage offset, and then moves it by the requested amount. The plugins themselves store their own offset, and they know how much storage they need, so EEPROMSettings doesn't need to.

This hook would be the query part that asks the plugins, since they have the info.

obra, 2022-07-11:
Mm. Yeah. I see. I feel like we *should* be able to tie them together more
tightly. I guess we could make a generic, reusable
Kaleidoscope::Plugin::requestSlice that uses onStorageSizeQuery to figure
out how much data to request. Doing this isn't *really* a blocker for this
landing, but I'd love to end up with an API where plugins need less code to
use EEPROM, rather than more.

closed 2022-07-21 by algernon