keyboardio/Kaleidoscope #1222: change Focus timeout behavior for friendlier interactive terminal use

tlyu, 2022-08-23:
> Larger than a feature request, this type of issue is for describing a proposed change to how Kaleidoscope works, in advance of a draft PR.

> What do you want to change?

Change the timeout behavior of Focus serial input handling. Right now, unbuffered interactive keystrokes result in each character being read as a separate Focus command. For each keystroke, Focus outputs a `.` on a line by itself due to being an unrecognized command.

Additionally, accept all combinations of CR and LF as end-of-line.

> How will it make Kaleidoscope better?

It will make it easier to interact with Focus using terminal programs that don't line-buffer their user input. Right now, it's difficult to use programs like `screen` and `tio` to interact with Focus plugins. Specialized interfaces like the Arduino Serial Console do work, but I think it's helpful to work with a wider variety of terminal programs.

> What trouble might users have in adapting to the new functionality?

It shouldn't affect most other use cases. I guess it's theoretically possible that someone has written software that depends on the current timeout behavior instead of sending a newline to delimit the command. I think that's unlikely, though.

> What trouble might developers have in adapting to the new functionality?

It shouldn't affect well-written software, and it might make it easier for developers to test new Focus plugins.

tlyu, 2022-08-25:
Update: I skimmed the diffs between v1.99.5 and master, and it looks like this might already be fixed? I'll need to figure out how to do a local build to test it out.

algernon, 2022-08-25:
From what I can tell, the fix *might* be a combination of 1d3c7b339926db05d1d7552c07e427df26c76c63, c9943057f4f490f55a4c2ded5d77a8babcbe3315 and 62f668fd25ec924b2d278360416125592a34b5c1, with 54fbb04eb6f0c610babfbd41492175919fb7e971 putting the crown on it.

tlyu, 2022-08-26:
Thanks! I tried the latest Kaleidoscope master, and the experience is somewhat improved, but still a bit lacking. It now seems to accept commands delimited with space, but not with LF. Line-buffered input still works, if I force that.

PR #1226 (2022-08-27): Focus: correctly handle delayed newline
Previously, a newline that wasn't read with the rest of the command in
a single event would get appended to the command buffer instead of
remaining in the "peek" buffer, so the command dispatch would never
fire.

Stash a space delimiter instead of storing it in the buffer, only to
overwrite it later.

Fix an off-by-one error that could cause a buffered command to not be
null-terminated. This probably didn't cause a problem in practice,
because Focus plugins mostly did a `strcmp` against a fixed string that
is smaller than the command buffer size.

Fixes  [... cut]
closed 2022-08-27 by obra