keyboardio/Kaleidoscope #1229: Compile error if NUMPAD is removed

akerl, 2022-09-01:
**Describe the bug**

I'm working with a custom sketch, and if I remove the NUMPAD layer, I get a seemingly unrelated error.

**To Reproduce**
1. Clone https://github.com/akerl/keyboard and check out the testing branch
2. `arduino-cli core install keyboardio:avr arduino:avr`
3. `arduino-cli compile --output-dir`

Get the following error:

```
/opt/build/keyboard.ino:58:38: error: variable or field 'hostPowerManagementEventHandler' declared void
 void hostPowerManagementEventHandler(kaleidoscope::plugin::HostPowerManagement::Event event) {
                                      ^~~~~~~~~~~~
/opt/build/keyboard.ino:58:38: error: 'kaleidoscope' has not been declared

Error during build: exit status 1
```

**Expected behavior**

Compile succeeds. For comparison:

1. Clone https://github.com/akerl/keyboard and check out the master branch
2. `arduino-cli core install keyboardio:avr arduino:avr`
3. `arduino-cli compile --output-dir`

**Environment (please complete the following information):**
 - OS: Archlinux
 - Device: Keyboard.io Model 01

obra, 2022-09-06:
I'm unable to reproduce this failure locally. (Sorry, I replied by email a few hours after the issue was opened, but it looks like it was swallowed up somewhere.)

Can you reproduce this issue making the same change to the "stock" model 01 firmware?

akerl, 2022-09-11:
Ok, so this is getting even stranger. I only get this error on Linux; it works when I build on OSX.

akerl, 2022-09-11:
Trying to build a test case based on the stock repo now.

obra, 2022-10-11:
Hi Les,

I'm unable to reproduce your failure locally if I check out your master
branch and make the identical changes to what's in your testing branch. For
that matter, your testing branch seems to work ok for me, too.

akerl, 2022-10-11:
Yea, I've narrowed this down to something weirder, where it builds fine on some of my systems but not others. I lost the motivation to chase it deeper, given that keeping the NUMPAD def in there makes it work everywhere. But it's safe to say I don't think this is an issue w/ the keyboardio code.

closed 2022-10-11 by akerl