keyboardio/Kaleidoscope #1235: Focus should be more resistant to random input / scans

dakkar, 2022-09-21:
**Describe the bug**
Programs that scan all connected serial/usb ports looking for some response, may send bytes that confuse Focus and (temporarily) lock up the keyboard. 

**To Reproduce**
For example: install Ultimaker's Cura and let it try to auto-detect a connected 3d printer. It runs https://github.com/Ultimaker/Cura/blob/main/plugins/USBPrinting/AutoDetectBaudJob.py sending a GCode command at different speeds, and expects a "ok\n" back.

A connected Atreus is reportedly unresponsive until Cura is terminated.

**Expected behavior**
Kaleidoscope/Focus should return some error message, without locking the keyboard.

**Additional context**
Work to fix this will probably intersect with work to fix #1211

tlyu, 2022-09-24:
I think 0.90.1 fixes enough of the underlying problems to prevent the keyboard from becoming unresponsive to keystrokes upon receiving an invalid command. Focus might still get out of sync enough to cause problems for Chrysalis.

tlyu, 2022-09-28:
@dakkar could you please check whether firmware 0.90.1 resolves the keyboard lockups caused by Cura?

dakkar, 2022-09-28:
I'll take a look on Friday… I don't actually use Cura, I just wrote this issue to capture a problem reported on Discord, but it's easy enough to install and test

dakkar, 2022-09-30:
bad news: it still needs work

I've built (and flashed) [my sketch](https://www.thenautilus.net/cgit/keyboardio-model01/tree/Model01-dakkar-firmware.ino) with:
* keyboardio/Kaleidoscope 9bf6eba010dd61fc56da377eda87aded43f4a8ad
* keyboardio/Kaleidoscope-Bundle-Keyboardio@7d68bb7745b61d2d9637b9513d06a2cca685b595
* keyboardio/ArduinoCore-GD32-Keyboardio@4fcce930d09ff9069649f17de9a497a9163b8f95

Then installed https://github.com/Ultimaker/Cura/releases/tag/5.1.1

With Cura open, the keyboard is unresponsive. More: Cura doesn't completely exit until the model100 is unplugged, probably because the event loop is still waiting for some response from the serial port.

dakkar, 2022-09-30:
I've cobbled together a minimal script that does the same thing as Cura w.r.t serial ports (as in, I've copied the code, removed the bits that depended on Ultimaker's other libraries, and put everything in a single file)

https://gist.github.com/dakkar/bb6b8ff269625b020405de8ad06df0d7

running this locks the keyboard the same way as Cura (as far as I can see)

program output:
```
starting
Auto detect baud rate started.
connecting as a programmer
failed, closing
Checking /dev/ttyACM0 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
Traceback (most recent call last):
  File "/tmp/cura-repro.py", line 209, in <module>
    adbj.run()
  File "/tmp/cura-repro.py", line 203, in run
    serial.write(b"M105\n")
  File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 636, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
```

dakkar, 2022-10-12:
Linux,  with:
* keyboardio/Kaleidoscope 8da824a8f88dfd8bd90b55b4beb93c1b3fe24806
* keyboardio/Kaleidoscope-Bundle-Keyboardio@7f6008abcc8ada3bca57897efa7520750a70d115
* keyboardio/ArduinoCore-GD32-Keyboardio@57fc0ec8c230d2dffa2a1482656f676db4ee075a

still locks up

tlyu, 2022-10-12:
> Linux, with:
> 
> * keyboardio/Kaleidoscope [8da824a](https://github.com/keyboardio/Kaleidoscope/commit/8da824a8f88dfd8bd90b55b4beb93c1b3fe24806)
> * [keyboardio/Kaleidoscope-Bundle-Keyboardio@7f6008a](https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio/commit/7f6008abcc8ada3bca57897efa7520750a70d115)
> * [keyboardio/ArduinoCore-GD32-Keyboardio@57fc0ec](https://github.com/keyboardio/ArduinoCore-GD32-Keyboardio/commit/57fc0ec8c230d2dffa2a1482656f676db4ee075a)
> 
> still locks up

Thanks. Is this the Model 100, or the Atreus? And by locking up, do you mean that the keyboard no longer sends keystrokes to the OS? Does it respond to the pressing the LED key to switch LED effects? If there are animated LED effects running, does the animation freeze?

Also, could you please confirm whether or not the lockup happens with the latest firmware from a fully-updated Chrysalis, instead of a custom sketch?

tlyu, 2022-10-12:
Confirmed in an Ubuntu 20.04 VM. I wasn't expecting it to reproduce in a VM, but apparently it does?

Anyway, connecting to the serial port with `screen` or `tio` after the lockup causes the lockup to end, and a series of `\r\n.\r\n` to be produced. I'm guessing that the output buffer filled up, and Linux doesn't drain pending serial input if nothing is reading it, unlike macOS.

tlyu, 2022-10-12:
I can semi-reproducibly produce a lockup by running
```sh
(stty raw -echo -F /dev/ttyACM0 ; echo; echo eeprom.contents; echo waiting > /dev/tty ; sleep 20)> /dev/ttyACM0
```
from a `bash` prompt in Linux.

tlyu, 2022-10-12:
More details: after adding some `print` statements to the Cura reproducer script, I noticed that it was repeatedly sending `M105\n` but only reading until `\n`, so the `\r\n.\r\n` kept backing up in the input queue because each unrecognized command produced one more newline than Cura was reading. This eventually produced a write timeout and keyboard lockup.

cxw42, 2024-03-10:
I am experimenting with https://github.com/cxw42/Kaleidoscope/commit/120a945ac15670c28e27f830475b86183a514b17 and it seems to be working OK so far!  When it gets an M105, it skips emitting the `\r\n.`.  Feedback welcome.  Kaleidoscope 1.99.8 except with FocusSerial.cpp changed.