keyboardio/Kaleidoscope #1237: Error code 74 when installing latest default firmware on my Model 100

jhurwitz, 2022-09-24:
**Describe the bug**
I've followed the [instructions](https://kaleidoscope.readthedocs.io/en/latest/build_default_firmware.html#build-the-firmware) to build and install the latest default firmware, as well as the [troubleshooting guide](https://github.com/keyboardio/Kaleidoscope/wiki/Troubleshooting-Firmware-Upload-Issues).

I cannot seem to upload the firmware to my Model 100. The full output from Arduino (with verbose output enabled) is:
```
Sketch uses 144400 bytes (13%) of program storage space. Maximum is 1048576 bytes.
Global variables use 21004 bytes (21%) of dynamic memory, leaving 77300 bytes for local variables. Maximum is 98304 bytes.
"/Users/jacob/Library/Arduino15/packages/keyboardio/tools/dfu-util/0.10.0-arduino1/dfu-util" --device 0x3496:0x0005 -D "/private/var/folders/v_/kn8w4x9j0z342xm5cfzwb1s80000gn/T/arduino-sketch-175BF70C3111598AA9389A44084BDC5F/Model100.ino.bin" -R 
dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available
Failed uploading: uploading error: exit status 74
```

**To Reproduce**
"Upload" the sketch while holding down the PROG key on my Model 100.

**Expected behavior**
The firmware is uploaded successfully.

**Environment (please complete the following information):**
 - macOS 12
 - Keyboardio Model 100
 - Arduino 2.0.0
 - repros on Kaleidoscope 1.99.5 and 2022.9.23193515

**Additional context**

ycros, 2022-09-24:
I'm getting the same error on Windows 11.
I've found a workaround, which is to hold prog while plugging the usb cable into the keyboard, and then flashing works fine.
Additionally, flashing 1.99.5 doesn't seem to work on the Model 100, but the latest versions work.

jhurwitz, 2022-09-24:
> I'm getting the same error on Windows 11.
> I've found a workaround, which is to hold prog while plugging the usb cable into the keyboard, and then flashing works fine.
> Additionally, flashing 1.99.5 doesn't seem to work on the Model 100, but the latest versions work.

Thanks for the tip. This worked for me as well.

obra, 2022-09-24:
The issue here is that Arduino doesn't know how to reboot the Model 100
into its bootloader. For now, if you want to use the IDE, you need to hold
in Prog while plugging in the keyboard to put it in bootloader mode.

jhurwitz, 2022-09-24:
Thanks Jesse. I've updated [the wiki](https://github.com/keyboardio/Kaleidoscope/wiki/Troubleshooting-Firmware-Upload-Issues) so that future Model 100 owners don't get stuck here.

closed 2022-09-24 by jhurwitz