keyboardio/Kaleidoscope #1244: model100 make flash failure: main file missing from sketch

ian-kelling, 2022-09-26:
**To Reproduce**

I ran the following, held down prog when it was time to flash.

```
$ export VERBOSE=1
$ export KALEIDOSCOPE_DIR=/a/opt/Kaleidoscope
$ cd /a/opt/Kaleidoscope/examples/Devices/Keyboardio/Model100
$ make compile
$ make flash
...

Kaleidoscope-Steno                        0.0.0   /a/opt/Kaleidoscope/plugins/Kaleidoscope-Steno
Wire                                      1.0     /a/opt/Kaleidoscope/.arduino/user/hardware/keyboardio/gd32/libraries/Wire

Used platform   Version Path
keyboardio:gd32 1.0.0   /a/opt/Kaleidoscope/.arduino/user/hardware/keyboardio/gd32

cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.hex" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3.hex"
cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.elf" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3.elf"
if [ -e "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.bin" ]; then cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.bin" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3.bin"; else :; fi
ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3.hex" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.hex"
ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3.elf" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.elf"
if [ -e "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3.bin" ]; then ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3.bin" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.bin"; else :; fi
If your keyboard needs you to do something to put it in flashing mode, do that now.

When you're ready to proceed, press 'Enter'.

ARDUINO_DIRECTORIES_USER=/a/opt/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/a/opt/Kaleidoscope/.arduino/data /a/opt/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port \
--input-dir "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino" \
--verbose
Error during Upload: main file missing from sketch: /tmp/kaleidoscope-iank/output/2522259519-Model100.ino/2522259519-Model100.ino.ino
make: *** [/a/opt/Kaleidoscope/etc/makefiles/sketch.mk:207: flash] Error 1

$ ll /tmp/kaleidoscope-iank/output/2522259519-Model100.ino
- 755  iank iank 155K 09-26 03:12 pm Model100.ino.bin
- 755  iank iank   5M 09-26 03:12 pm Model100.ino.elf
- 644  iank iank 434K 09-26 03:12 pm Model100.ino.hex
- 644  iank iank   6M 09-26 03:12 pm Model100.ino.map
- 755  iank iank 154K 09-26 03:04 pm Model100-keyboardio-m100-MP-1-34-gf06624.bin
- 755  iank iank   5M 09-26 03:04 pm Model100-keyboardio-m100-MP-1-34-gf06624.elf
- 644  iank iank 434K 09-26 03:04 pm Model100-keyboardio-m100-MP-1-34-gf06624.hex
- 755  iank iank 155K 09-26 03:12 pm Model100-keyboardio-m100-MP-1-35-gbe28b3.bin
- 755  iank iank   5M 09-26 03:12 pm Model100-keyboardio-m100-MP-1-35-gbe28b3.elf
- 644  iank iank 434K 09-26 03:12 pm Model100-keyboardio-m100-MP-1-35-gbe28b3.hex
l 777  iank iank   44 09-26 03:12 pm Model100-latest.bin -> Model100-keyboardio-m100-MP-1-35-gbe28b3.bin
l 777  iank iank   44 09-26 03:12 pm Model100-latest.elf -> Model100-keyboardio-m100-MP-1-35-gbe28b3.elf
l 777  iank iank   44 09-26 03:12 pm Model100-latest.hex -> Model100-keyboardio-m100-MP-1-35-gbe28b3.hex
```

**Expected behavior**

flash of firmware happens

**Environment (please complete the following information):**
 - OS: trisquel 11 (for this purpose, it should work the same as ubuntu
   22.04, unless some os-detection went wrong)
 - Device: Keyboardio Model 100

```
git log -1
commit be28b367cee2a016c147477a782cbaefc345f9ca (HEAD -> master, origin/master, origin/HEAD)
Merge: a9f4f16e 45bd9d1a
Author: Jesse Vincent <jesse@fsck.com>
Date:   Mon Sep 26 11:30:58 2022 -0700

    Merge pull request #1243 from tlyu/fix-gamma

    restore gamma correction on Model 100
```

**Additional context**

This is my first attempt at flashing after receiving the keyboard using
an unmodified repository.

ian-kelling, 2022-09-26:
I realized I had run `make setup` long ago in this repository, and perhaps that had left something around even though I ran it again after pulling. Sure enough, after doing `git clean -xfffd`, then starting from `make setup`, this error disappeared. I have a new one though, I'll open a different bug.

ian-kelling, 2022-09-26:
Actually, I fixed the other error and this one came back. So, reopening.

obra, 2022-09-26:
Can you try rerunning this after also exporting `export QUIET=`

(That is, setting the variable, but setting it to nothing)

ian-kelling, 2022-09-26:
yes. now, it looks like the other error I had

```
$ make flash
...
cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.hex" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.hex"
cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.elf" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.elf"
if [ -e "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.bin" ]; then cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.bin" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.bin"; else :; fi
ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.hex" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.hex"
ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.elf" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.elf"
if [ -e "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.bin" ]; then ln -sf "Model100-keyboardio-m100-MP-1-35-gbe28b3-dirty.bin" "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100-latest.bin"; else :; fi
To update your keyboard's firmware, hold down the 'Prog' key on your keyboard.  (When flashing starts, you can release it.)

When you're ready to proceed, press 'Enter'.

ARDUINO_DIRECTORIES_USER=/a/opt/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/a/opt/Kaleidoscope/.arduino/data /a/opt/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port /dev/ttyACM0 \
--input-dir "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino" \
--verbose
"/a/opt/Kaleidoscope/.arduino/data/packages/keyboardio/tools/dfu-util/0.10.0-arduino1/dfu-util" --device 0x3496:0x0005 -D "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100.ino.bin" -R 
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: No DFU capable USB device available
Error during Upload: Failed uploading: uploading error: exit status 74
make: *** [/a/opt/Kaleidoscope/etc/makefiles/sketch.mk:205: flash] Error 1

```

ian-kelling, 2022-09-26:
I'm investigating this error, I may be able to figure it out.

ian-kelling, 2022-09-26:
with model 100 plugged in

```
$ lsusb|grep -i keybo
Bus 001 Device 027: ID 3496:0006 Keyboardio Model 100
```
with model 01 plugged in:
```
$ lsusb|grep -i keybo
Bus 001 Device 028: ID 1209:2301 Generic Keyboardio Model 01
```

that 2301 number is mentioned here https://raw.githubusercontent.com/keyboardio/Kaleidoscope/master/etc/60-kaleidoscope.rules, per https://kaleidoscope.readthedocs.io/en/latest/setup_toolchain.html, looks like the Model100 needs to be added.

obra, 2022-09-26:
...that sure could do it.

Thank you.

tlyu, 2022-09-26:
> yes. now, it looks like the other error I had
>
> ```
> $ make flash
> ...
> cp "/tmp/kaleidoscope-iank/build/2522259519-Model100.ino/Model100.ino.hex" [...]

That looks like it's failing to reset the keyboard before running `arduino-cli upload`. As a workaround, you might try unplugging and replugging the Model 100 with the PROG key held down when you're prompted to press Enter, then press Enter using a different keyboard.

ian-kelling, 2022-09-26:
I've determined that the first error happens when the keyboard is not detected by the OS. I've found plugging it in doesn't always work, replugging fixed is. I was using my laptop keyboard to do input, so I didn't notice.

Ok, I created some udev lines based on the other ones, and got further. I can't tell if this error means the flash didn't succeed. I will have to test making a modification and seeing if it shows up. This might belong in another bug.

```
When you're ready to proceed, press 'Enter'.

DEVICE=/dev/ttyACM0 /a/opt/Kaleidoscope/bin/focus-send "device.reset"
sleep 2
ARDUINO_DIRECTORIES_USER=/a/opt/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/a/opt/Kaleidoscope/.arduino/data /a/opt/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port /dev/ttyACM0 \
--input-dir "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino" \
--verbose
"/a/opt/Kaleidoscope/.arduino/data/packages/keyboardio/tools/dfu-util/0.10.0-arduino1/dfu-util" --device 0x3496:0x0005 -D "/tmp/kaleidoscope-iank/output/2522259519-Model100.ino/Model100.ino.bin" -R 
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 3496:0005
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [=========================] 100%       158792 bytes
Download done.
DFU state(7) = dfuMANIFEST, status(0) = No error condition is present
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to Run-Time mode
Error during Upload: Failed uploading: uploading error: exit status 251
make: *** [/a/opt/Kaleidoscope/etc/makefiles/sketch.mk:207: flash] Error 1

```

obra, 2022-09-26:
That's a bug in dfu-util which should be fixed when we push the next bsp
rev. The update succeeded, but that version of dfu incorrectly exits with
an error code.

ian-kelling, 2022-09-26:
Note: I just unplugged and replugged it and my other keyboard about 10 times with the cable that came in the box and didn't reproduce the keyboard not being detected by the OS, so perhaps it sorted itself out.

ian-kelling, 2022-09-26:
Excellent. I would make a pull request for the udev rule, but I don't know exactly what the full set of rules should be be since the existing rules have 2 idProduct numbers per model, but I only get 1 from lsusb.

tlyu, 2022-09-26:
> Excellent. I would make a pull request for the udev rule, but I don't know exactly what the full set of rules should be be since the existing rules have 2 idProduct numbers per model, but I only get 1 from lsusb.

That is expected. There's one PID for the bootloader, and another for the normal device. Only one of them will appear at once. It's probably easiest to sync with the version of the udev rules in the Chrysalis repository.

neh, 2022-09-26:
I'm getting the exact same output as @ian-kelling with exit status 251 (in https://github.com/keyboardio/Kaleidoscope/issues/1244#issuecomment-1258616764) and the update does not seem to be succeeding. The changes I make in the Model100.ino file are not taking effect. Making changes and flashing from Chrysalis does work, though.

tlyu, 2022-09-26:
> Making changes and flashing from Chrysalis does work, though.

What specific changes did you make, and how did you confirm that they aren't taking effect?

neh, 2022-09-26:
In Chrysalis I changed a few thumb keys around just to check that flashing was working, and it did. Then I ran `make flash` with the example Model100.ino without making any changes to it, just to reset to defaults, and the changes I made in Chrysalis were still in effect. I also just made a few letter keys in the .ino all the same, but they stayed as they were before.

tlyu, 2022-09-26:
> In Chrysalis I changed a few thumb keys around just to check that flashing was working, and it did. Then I ran `make flash` with the example Model100.ino without making any changes to it, just to [...]

I think that is expected behavior. Chrysalis stores key layouts in EEPROM, which isn't usually reset when flashing from a sketch. The EEPROM layouts override the ones compiled into the firmware. I think you would have to do a factory reset in Chrysalis to see the layout changes from the firmware changes.

neh, 2022-09-26:
I did try a factory reset previously, but I just did it again and now it's working. Thank you for your help and sorry for the derailment.

tlyu, 2022-09-26:
> I did try a factory reset previously, but I just did it again and now it's working. Thank you for your help and sorry for the derailment.

No worries! This behavior is admittedly somewhat surprising and should probably be better documented.

neh, 2022-09-26:
It probably should be... I now vaguely recall running into the exact same issue a few years ago when setting up my Atreus.

tlyu, 2022-09-26:
@ian-kelling #1245 should update the copy of the udev rules in the Kaleidoscope repository.

ian-kelling, 2022-09-26:
Thanks! I'm resolving this then.

closed 2022-09-26 by ian-kelling

tsibley, 2022-09-27:
FWIW, I get this same error:

```
Error during Upload: main file missing from sketch: /tmp/kaleidoscope-tom/output/382451299-Model100.ino/382451299-Model100.ino.ino
```

when running `make flash` from current master when the Model 100 is in bootloader mode (i.e. I held Prog while plugging it in):

```
$ lsusb -d 3496:
Bus 001 Device 006: ID 3496:0005 Keyboardio Model 100 (Bootloader)
```

tsibley, 2022-09-27:
If instead start with it out of bootloader mode:

```
$ lsusb -d 3496:
Bus 001 Device 010: ID 3496:0006 Keyboardio Model 100
```

and run `make flash`, hold Prog, hit enter, etc. then the firmware is uploaded and works, but ultimately `make` still fails with:

```
Resetting USB to switch back to Run-Time mode
Error during Upload: Failed uploading: uploading error: exit status 251
```
<details>
<summary>Details</summary>

```
DEVICE=/dev/ttyACM0 /home/tom/projects/Kaleidoscope/bin/focus-send "device.reset"
sleep 2
ARDUINO_DIRECTORIES_USER=/home/tom/projects/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/home/tom/projects/Kaleidoscope/.arduino/data /home/tom/projects/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port /dev/ttyACM0 \
--input-dir "/tmp/kaleidoscope-tom/output/382451299-Model100.ino" \
--verbose
"/home/tom/projects/Kaleidoscope/.arduino/data/packages/keyboardio/tools/dfu-util/0.10.0-arduino1/dfu-util" --device 0x3496:0x0005 -D "/tmp/kaleidoscope-tom/output/382451299-Model100.ino/Model100.ino.bin" -R 
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 3496:0005
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        99452 bytes
Download done.
DFU state(7) = dfuMANIFEST, status(0) = No error condition is present
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to Run-Time mode
Error during Upload: Failed uploading: uploading error: exit status 251
make: *** [/home/tom/projects/Kaleidoscope/etc/makefiles/sketch.mk:207: flash] Error 1

```

</details>

ian-kelling, 2022-09-27:
@tsibley , as we said in earlier comments, even though it failed with 251, it actually succeeded.

tsibley, 2022-09-27:
@ian-kelling Indeed. That’s what I meant by “then the firmware is uploaded and works”. Sorry if that wasn’t clear. I commented here only because I thought it curious that the “main file missing” error was associated with the device already being in bootloader mode (vs. not).