keyboardio/Kaleidoscope #1272: Model 100 keymap appears scrambled in Chrysalis after firmware update

blm768, 2022-10-13:
**Describe the bug**
After [this commit](https://github.com/keyboardio/Kaleidoscope/commit/b949ca89b1ebfaf5018dac4ee2b2febdb5554c4b) (or somewhere nearby, but definitely after [this one](https://github.com/keyboardio/Kaleidoscope/commit/9bf6eba010dd61fc56da377eda87aded43f4a8ad)), compiling and flashing the firmware leaves my Model 100 in a state where everything in the keymap appears shifted over several columns in Chrysalis. I haven't had the chance to thoroughly test the actual key behavior yet, but it's definitely wrong if I reload the stock QWERTY layout and try to flash that to EEPROM. It looks like Chrysalis (or the firmware) is just confused as to which keymap slots map to actual physical keys.

**To Reproduce**

1. Check out the affected commit. (My Arduino setup is a bit broken at the moment, so I'm having trouble isolating the specific commit, but I can try to debug further on another machine when I have more time.)
2. Run `make flash` from the Model100 directory.
3. Attach to the keyboard in Chrysalis

**Expected behavior**
I'd expect my keymap to appear the same in Chrysalis as it did with the previous firmware.

**Environment (please complete the following information):**
 - OS: NixOS
 - Version: 22.05
 - Device: Keyboardio Model 100

**Additional context**
I'd probably better test this on my Windows machine just in case something's weird with my NixOS setup, so don't spend a ton of time trying to track this down, especially if it doesn't reproduce easily.

Kethku, 2022-10-18:
I'm also encountering this issue with the model 100 I just recieved.

blm768, 2022-10-24:
I think Pull Request #1291 may be enough to mitigate this.

closed 2022-10-24 by blm768