keyboardio/Kaleidoscope #1278: Can't get make flash to work on Win11

forrcaho, 2022-10-16:
I am trying to flash my Model100 from a working copy of this github repo. I'm on Windows 11 and I am using a mingw64 build environment.

Although I had to run `make setup` under Powershell, mingw64 works for the compile step. The `make flash` step only dies at the very end. Here is the last bit of output:

```
cp "/tmp/kaleidoscope-forrcaho/build/1653700070-Model100.ino/Model100.ino.hex" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-v1.99.7-4-gd745b4.hex"
cp "/tmp/kaleidoscope-forrcaho/build/1653700070-Model100.ino/Model100.ino.elf" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-v1.99.7-4-gd745b4.elf"
if [ -e "/tmp/kaleidoscope-forrcaho/build/1653700070-Model100.ino/Model100.ino.bin" ]; then cp "/tmp/kaleidoscope-forrcaho/build/1653700070-Model100.ino/Model100.ino.bin" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-v1.99.7-4-gd745b4.bin"; else :; fi
ln -sf "Model100-v1.99.7-4-gd745b4.hex" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-latest.hex"
ln -sf "Model100-v1.99.7-4-gd745b4.elf" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-latest.elf"
if [ -e "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-v1.99.7-4-gd745b4.bin" ]; then ln -sf "Model100-v1.99.7-4-gd745b4.bin" "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100-latest.bin"; else :; fi
To update your keyboard's firmware, hold down the 'Prog' key on your keyboard.  (When flashing starts, you can release it.)

When you're ready to proceed, press 'Enter'.

# Earlier Model 100 firmware didn't implement 1200bps touch
# If we have a device serial port available, try to trigger a Kaliedoscope reset
export DEVICE= && \
        [ -e "$DEVICE" ] && \
        /c/Users/forrcaho/Kaleidoscope/bin/focus-send "device.reset" && \
        sleep 2
make: [/c/Users/forrcaho/Kaleidoscope/etc/makefiles/sketch.mk:206: flash] Error 1 (ignored)
ARDUINO_DIRECTORIES_USER=/c/Users/forrcaho/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/c/Users/forrcaho/Kaleidoscope/.arduino/data /c/Users/forrcaho/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port \
--input-dir "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino" \
--verbose
Error during Upload: main file missing from sketch: C:\msys64\tmp\kaleidoscope-forrcaho\output\1653700070-Model100.ino\1653700070-Model100.ino.ino
make: *** [/c/Users/forrcaho/Kaleidoscope/etc/makefiles/sketch.mk:207: flash] Error 1
```

Note that it is expecting to find the file `1653700070-Model100.ino.ino` in the directory `C:\msys64\tmp\kaleidoscope-forrcaho\output\1653700070-Model100.ino`. That directory exists and seems to have a lot of relevant files:

```
 $ ls  -al /c/msys64/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/
total 24196
drwxr-xr-x 1 forrcaho None       0 Oct 15 22:28 .
drwxr-xr-x 1 forrcaho None       0 Oct 15 22:18 ..
-rw-r--r-- 1 forrcaho None  159496 Oct 15 22:28 Model100.ino.bin
-rw-r--r-- 1 forrcaho None 5378468 Oct 15 22:28 Model100.ino.elf
-rw-r--r-- 1 forrcaho None  448700 Oct 15 22:28 Model100.ino.hex
-rw-r--r-- 1 forrcaho None 6793300 Oct 15 22:28 Model100.ino.map
-rw-r--r-- 1 forrcaho None  159496 Oct 15 22:28 Model100-latest.bin
-rw-r--r-- 1 forrcaho None 5378468 Oct 15 22:28 Model100-latest.elf
-rw-r--r-- 1 forrcaho None  448700 Oct 15 22:28 Model100-latest.hex
-rw-r--r-- 1 forrcaho None  159496 Oct 15 22:28 Model100-v1.99.7-4-gd745b4.bin
-rw-r--r-- 1 forrcaho None 5378468 Oct 15 22:28 Model100-v1.99.7-4-gd745b4.elf
-rw-r--r-- 1 forrcaho None  448700 Oct 15 22:28 Model100-v1.99.7-4-gd745b4.hex
```

There is, however, no `.ino.ino` file.

I have been able to flash my own code from the Arduino IDE on Win11, so I know that part works.

forrcaho, 2022-10-18:
After getting the latest contents of master and running `make update` I'm now getting a different error from `make flash`:

```
To update your keyboard's firmware, hold down the 'Prog' key on your keyboard.  (When flashing starts, you can release it.)

When you're ready to proceed, press 'Enter'.

# Earlier Model 100 firmware didn't implement 1200bps touch
# If we have a device serial port available, try to trigger a Kaleidoscope reset
export DEVICE=COM7 && \
        [ -e "$DEVICE" ] && \
        /c/Users/forrcaho/Kaleidoscope/bin/focus-send "device.reset" && \
        sleep 2
make: [/c/Users/forrcaho/Kaleidoscope/etc/makefiles/sketch.mk:206: flash] Error 1 (ignored)
ARDUINO_DIRECTORIES_USER=/c/Users/forrcaho/Kaleidoscope/.arduino/user ARDUINO_DIRECTORIES_DATA=/c/Users/forrcaho/Kaleidoscope/.arduino/data /c/Users/forrcaho/Kaleidoscope/bin/arduino-cli upload --fqbn keyboardio:gd32:keyboardio_model_100 \
--port COM7 \
--input-dir "/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino" \
--verbose
Performing 1200-bps touch reset on serial port COM7
TOUCH: error during reset: opening port at 1200bps: Invalid serial port
"C:\Users\forrcaho\Kaleidoscope\.arduino\data\packages\keyboardio\tools\dfu-util\0.10.0-arduino1/dfu-util" --device 0x3496:0x0005 -D "C:/msys64/tmp/kaleidoscope-forrcaho/output/1653700070-Model100.ino/Model100.ino.bin" -R
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release
No DFU capable USB device available
Error during Upload: Failed uploading: uploading error: exit status 74
make: *** [/c/Users/forrcaho/Kaleidoscope/etc/makefiles/sketch.mk:207: flash] Error 1
```
COM7 is also where Chrysalis sees my keyboard, so I think that's correct.

forrcaho, 2022-10-18:
For anyone else who is having this problem, I was able to flash the code I compiled with `make compile` to my Model 100 using Chrysalis, following these instructions from algernon on discord:

> make `OUTPUT_PATH=/some/path`, and the compiled firmware will be under `/some/path`. Then you can use Chrysalis to flash it, by going to Firmware Update, selecting "Custom firmware", and select the file ending with `-latest.bin` from `/some/path`. Then proceed with flashing through Chrysalis!

I'm leaving this issue open since I still can't flash using `make flash`, but now that I have a way to flash successfully, it's not a critical issue for me. Feel free to close it.

obra, 2022-10-20:
I *believe* that with last night's changes to Kaleidoscope, this `make flash` issue should now be resolved. I'm closing out this ticket, but please do reopen it if things are still misbehaving.

closed 2022-10-20 by obra