keyboardio/Kaleidoscope #1285: Focus parsing changes

tlyu, 2022-10-18:
Larger than a feature request, this type of issue is for describing a proposed change to how Kaleidoscope works, in advance of a draft PR.

## What do you want to change?

Move all parsing of Focus input into `FocusSerial` itself. This is an incremental improvement to Focus. Further design improvements might be desired in the future.

There are probably no changes required to the protocol syntax for now. Improvements to the protocol syntax can be handled as a separate issue.

### Details

#### Data types

Focus will define an enumerated list of data types that plugins may parse from commands. There will be a null type to indicate that no further items are expected.

#### Command table

Each Focus plugin will define a command table. Each entry will contain (at minimum) a string to invoke the command. Upon receiving the start of a command over the serial input, Focus will scan the command table of each plugin for a matching command. Upon detecting a match, Focus will call a plugin method (name TBD, possibly `onFocusBeginCommand`) with the index of the matching command table entry. The method will return the data type of the next expected item.

One advantage of using command tables is that plugins are no longer responsible for matching input strings to determine if they have received a valid command. This eliminates a lot of repetitive code in plugins.

Another advantage of using command tables is that Focus can automatically generate the help output without further input from the plugin.

#### Per-item callbacks

Each time Focus parses another item from the serial input, it will call the appropriate callback for the data type requested by the plugin. Each plugin will return the data type of the next expected item.

#### End of command

Upon reading EOL from the serial input, Focus will call a plugin method (name TDB, possibly `onFocusEndCommand`) to indicate that the command is complete. Focus will write `\r\n.\r\n` to serial output, as currently, to indicate end the of command output.

#### Parse errors

Focus will call a plugin method for a parse error encountered while reading an item (name TBD, possibly `onFocusParseError`).

#### Command output

Any command output can be written using the same techniques as currently.

### Backward compatibility

Focus could continue to call `onFocusEvent` for backward compatibility with older plugins.

## How will it make Kaleidoscope better?

- Mitigate lockups due to malformed input (e.g., from probing by third-party software).

- Mitigate settings corruption caused by timing quirks from software (chunked writes, etc.).

- Lockups due to unread output from Focus are a separate issue.

## What trouble might users have in adapting to the new functionality?

None, unless there are bugs.

## What trouble might developers have in adapting to the new functionality?

This requires adapting plugins to a callback-based flow for reading commands. Therefore, each Focus plugin needs to keep additional state while a command is being parsed. This additional state might include the index of the command being executed, as well as loop counters for commands that accumulate array-like input.

## Alternatives considered

Having the command table declare the format of the entire command is a possibility, allowing Focus to accumulate an entire buffer of parsed input to pass to the plugin. Unfortunately, this might occupy too much RAM for AVR.

obra, 2022-10-18:
In general, this sounds pretty good to me. I'd love to explore what the ram overhead would look like for having the command table declare the format of the entire command.  My first gut reaction is that if we can make it go, it'll result in cleaner, simpler code.

tlyu, 2022-10-18:
Yeah, I was thinking that `eeprom.contents` as a write command would probably be the most extreme case (and one that's probably rarely used).

obra, 2022-10-18:
I'm not entirely convinced that eeprom.contents as a write command is
actually generally useful.

tlyu, 2022-10-18:
> Would the command table declaring the entire format require us to allocate memory for the parsed input of all commands, together? Because that basically means we need to have enough RAM free to hold a copy of eeprom - or at least, as much RAM as the number of bytes we use from EEPROM.

It would mean that we would need to at least dynamically allocate enough RAM to hold the parsed arguments of the command with the longest arguments.

obra, 2022-10-18:
> Can't we statically allocate enough for the longest? Dynamic allocation on
> AVR is *expensive*.
>
> Probably. We'd need to have a way to make sure it's big enough.

I believe the current design was "a first pass" more than anything else.

I'd be thrilled for us to migrate to commands that operate on a single
keymap at a time.

tlyu, 2022-10-19:
I think we can leave some trickier commands like `eeprom.contents` using the legacy API, at least to start with.