keyboardio/Kaleidoscope #1295: Key with secondary action to move to another layer ends up repeating instead of moving to the layer

hwangdylan, 2022-10-24:
i have this space key setup with a secondary action to shift to another layer when held. Thus the expected behavior is that when you press the key and hold it will shift to the other layer. However what i've noticed is that if i press and hold that space key very soon after pressing an actual character, it will actually start repeating the previous key pressed. so sometimes this means that what i type out becomes 'howwwwwwwww are you ' instead of "how are you". if i press and hold the space key in quick succession after pressing another key, it sometimes also just starts repeating spaces like "f                                ". I have tried this on my Atreus and gets the same issue so it seems agnostic to the actual device. 
[image]

gedankenexperimenter, 2022-10-24:
The plugin responsible for handling "secondary actions" is Qukeys.  It employs several mechanisms to avoid unintended modifiers, which generally lead to worse outcomes than unintended printable characters.  One of those mechanisms is the "minimum prior interval", which is the minimum amount of time that must elapse between the press of any key and the press of the "secondary action" key for that key to be eligible to take on it's alternate value (i.e. "secondary action").  The default for this interval is 75 milliseconds.  This explains the latter behaviour you mentioned, where you get repeating spaces instead of the desired layer shift.

The repeat of the prior key is more perplexing, however, and  seems like a genuine bug to me.  I assume that in the "how are you" example, you're strictly "tapping" keys, and none of them are held for long enough to trigger repeats (i.e. more than half a second on most systems).  If you were only seeing this on a Model100, and not an Atreus, I would guess that it was another undesirable interaction with SpaceCadet being (essentially) out of order in the stock firmware.  However, the Atreus sketch should not have any such problems in that regard.  If, however, you're using a custom firmware on your keyboards, it would be helpful to see the relevant sketch(es).

So far, I have no hypothesis for the cause of this bug.  It might help to know how fast you type to get an idea of the timing we might need to use to reproduce it in the simulator.

gedankenexperimenter, 2022-10-24:
On second thought, if my supposition is incorrect about the timing in the "howwwww are you" example, and you are holding the `space` key for a while after pressing `W`, that probably makes sense, because Qukeys doesn't usually re-order events.  I'll need to check the code to be sure, but if it doesn't, then a reasonable argument could be made that it should in this case, allowing the release event to skip the queue in order to avoid waiting on resolution of a qukey that was pressed after it.  We would only want to do this if the key in question was not a modifier or layer shift key, however.

hwangdylan, 2022-10-24:
So I just tested it out again with the atreus, seems like the "howwwww" example can not be replicated on the atreus, only the "f               " example is able to be replicated. However the other statements with the model 100 stand true. 

RE: 75 ms interval for secondary action. Could you explain a bit more as to why the repeated spaces is expected? I can understand why it would result in one space being pressed but I don't quite understand why it would result in repeated spaced. In terms of a user perspective, I think it's fair to say that If i were to press the space key as I'm typing, it would be reasonable to expect that it at the very most it would result in no space and instead shift the user to the layer. Repeated spaces is a pretty unexpected behavior. 

As for typing speed, my normal typing speed is probably around 80 in normal cases.

gedankenexperimenter, 2022-10-24:
If the one example only appears on the Model100, that's probably because SpaceCadet and Qukeys are initialized in the wrong order in that sketch.

In the other case, I expect `f       ` if you press <kbd>F</kbd>, press <kbd>space</kbd> less than 75 ms later, release <kbd>F</kbd>, and hold <kbd>space</kbd> for long enough for the OS repeat to start (probably >500 ms).