keyboardio/Kaleidoscope #1318: Model 100 shift key behavior broken after firmware update

khers, 2022-12-11:
**Describe the bug**
My shift keys now act as parenthesis keys when I do not pair them with another key.  Also, if I am holding a key down (say 'w') and want to change to 'W' with shift this no longer works and i get a left or right parenthesis on release.

**To Reproduce**
Steps to reproduce the behavior:
Update Model 100 to latest firmware
Play a game using wasd to move and try to run by holding shift. 

**Expected behavior**
Shift should toggle on press between w and W but instead I get:
wwwwwwwwwwwwwwwwwwwwwwwwwwwwwww(

**Environment (please complete the following information):**
 - OS:Pop_OS!
 - Version  22.04
 - Device Model 100

**Additional context**
I only updated the firmware to see if it would fix another bug I was seeing where if a key was pressed when the computer was off or asleep and then the computer was started no key presses would register until I replugged it.

obra, 2022-12-11:
Hi Eric,

This is a bug in Chrysalis - it accidentally enabled Space Cadet shift on
you. If you go into Chrysalis settings and disable Space Cadet Shift, you
should be good to go.

best,
jesse

khers, 2022-12-11:
Thanks Jesse,

I have not yet been able to get Chrysalis running, it isn't packaged for Debian/Ubuntu and I cannot find installation instructions that work (I opened request #1239 about it).

Eric

khers, 2022-12-11:
I commented out SpaceCadet and the Config from the init plugins list and this is fixed.

closed 2022-12-11 by khers